pierremolinaro/acan2517FD #1: Configuration error 0x20000

ferasterkawi, 2019-02-02:
Goood job... Thanks a lot Mr.Pierremolinaro #^_^#

I'm trying to send & receive sensors data between tow nodes ESP32 in FD mode.
I don't know how to append a specific data to frame when it declared: 
CANFDMessage frame ;
when I uploaded the same example sketch to each ESP, serial monitor appears this lines without any received frame: 

sizeof (ACAN2517FDSettings): 44 bytes
Configure ACAN2517FD
MCP2517FD RAM Usage: 2016 bytes
Configuration error 0x20000
Sent: 1
Sent: 2
Sent: 3
Sent: 4
Sent: 5
Sent: 6
Sent: 7
Sent: 8
Sent: 9

and continues sending...

ferasterkawi, 2019-02-02:
Hello... Thanks for help.. #^_^#
I changed the quartz & it is Receiving now...

But I tried to send one byte from the first node:
 CANFDMessage frame ;
 frame.id = 0x001 ; // Set the message identifier (it is 0 by default)
 frame.ext = true ;  // message is an extended one (it is a standard one by default)
 frame.len = 1 ; //One byte
 frame.data [0] = 0x12 ; // First data byte is 0x12

The second node received 0:

sizeof (ACAN2517FDSettings): 44 bytes
Configure ACAN2517FD
MCP2517FD RAM Usage: 2016 bytes
Bit Rate prescaler: 1
Arbitration Phase segment 1: 255
Arbitration Phase segment 2: 64
Arbitration SJW:64
Actual Arbitration Bit Rate: 125000 bit/s
Exact Arbitration Bit Rate ? yes
Arbitration Sample point: 80%
Sent: 1
Received: 1
Byte Received: 0
Sent: 2
Received: 2
Byte Received: 0
Sent: 3
Received: 3
Byte Received: 0
...

Are there any recommends to solve this problem???

pierremolinaro, 2019-02-03:
Hello,

I need you to give me your code, I can't help you without knowing it.

Perhaps you have still the line:
 	settings.mRequestedMode = ACAN2517FDSettings::LoopBackMode ; // Select loopback mode

If so, the MCP2517FD is in loopback mode and does not incoming messages from the network.

Pierre Molinaro

ferasterkawi, 2019-02-03:
Hello Mr. Molinaro
Thanks a lot for your help.. I removed that line & it is working nowww... #^_^#

closed 2019-02-06 by pierremolinaro

sidec1993, 2019-06-04:
hello, 
I am trying to configure MCP2517FD in NormalFD mode without using INT. Message sending is working perfectly as explained but message is not received on the receiver node. I am using receive() method for message receive. I am using CANFD with 40Mhz crystal. here is my code can you tell me where I am doing mistake. is it necessary to use INT and interrupt service routine for communication between two nodes.
-------I am getting this output-------

sizeof (ACAN2517FDSettings): 39 bytes
Configure ACAN2517FD
MCP2517FD RAM Usage: 2016 bytes
init buffers
16
begin
allocate
end
32
begin
allocate
end
Reset RAM
Bit Rate prescaler: 1
Arbitration Phase segment 1: 31
Arbitration Phase segment 2: 8
Arbitration SJW:8
Actual Arbitration Bit Rate: 1000000 bit/s
Exact Arbitration Bit Rate ? yes
Arbitration Sample point: 80%
Sent: 1
Sent: 2
Sent: 3
Sent: 4
Sent: 5
Sent: 6
Sent: 7
Sent: 8
Sent: 9
Sent: 10
Sent: 11
--------------------------------

-----------CODE-----------

#include <ACAN2517FD.h>
#include <ACAN2517FDSettings.h>
#include <CANFDMessage.h>
#include <ACAN2517FDFilters.h>
#include <SPI.h>

static const byte MCP2517_CS  = 10 ; // CS input of MCP2517
static const byte MCP2517_INT =  2 ; // INT output of MCP2517

// Driver object

ACAN2517FD can (MCP2517_CS, SPI ,255) ; //CHANGE

//   SETUP

void setup ()
{

  

  //--- Start serial
  Serial.begin (115200) ;
  //--- Wait for serial (blink led at 10 Hz during waiting)
  while (!Serial) {
    delay (50) ;

    //----------------------------------- Begin SPI

}   
 
  SPI.begin () ;

  //--- Configure ACAN2517FD
  Serial.print ("sizeof (ACAN2517FDSettings): ") ;
  Serial.print (sizeof (ACAN2517FDSettings)) ;
  Serial.println (" bytes") ;
  Serial.println ("Configure ACAN2517FD") ;
  ACAN2517FDSettings settings (ACAN2517FDSettings:: OSC_40MHz , 1000UL * 1000UL, ACAN2517FDSettings::DATA_BITRATE_x5) ; // ADDED
 settings.mRequestedMode = ACAN2517FDSettings::NormalFD;// InternalLoopBack ;  
  //--- Default values are too high for an Arduino Uno that contains 2048 bytes of RAM: reduce them

  ACAN2517FDSettings::PAYLOAD_8;
  settings.mDriverTransmitFIFOSize = 16 ;  
  settings.mControllerTransmitFIFOSize = 1;
  settings.mControllerReceiveFIFOSize = 16;
  settings.mDriverReceiveFIFOSize = 32 ;  
  settings.mControllerTXQSize = 11 ;
 
  //settings.mControllerTXQBufferPayload = PAYLOAD_8 
 
 //--- RAM Usage
  
  Serial.print ("MCP2517FD RAM Usage: ") ;
  Serial.print (settings.ramUsage ()) ;
  Serial.println (" bytes") ;
  
  //--- Begin
  
/*  ACAN2517FDFilters filters ;
  ACANFDCallBackRoutine receiveFromFilter0; 
  filters.appendFrameFilter (kStandard , 0x123 , receiveFromFilter0) ; 
 const uint32_t errorCode = can.begin(settings,NULL  ) ;

  
  if (errorCode == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Arbitration Phase segment 1: ") ;
    Serial.println (settings.mArbitrationPhaseSegment1) ;
    Serial.print ("Arbitration Phase segment 2: ") ;
    Serial.println (settings.mArbitrationPhaseSegment2) ;
    Serial.print ("Arbitration SJW:") ;
    Serial.println (settings.mArbitrationSJW) ;
    Serial.print ("Actual Arbitration Bit Rate: ") ;
    Serial.print (settings.actualArbitrationBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact Arbitration Bit Rate ? ") ;
    Serial.println (settings.exactArbitrationBitRate () ? "yes" : "no") ;
    Serial.print ("Arbitration Sample point: ") ;
    Serial.print (settings.arbitrationSamplePointFromBitStart ()) ;
    Serial.println ("%") ;
  }
  else
  {
    Serial.print ("Configuration error 0x") ;
    Serial.println (errorCode, HEX) ;
  }
}

//   LOOP

static uint32_t gSendDate = 0 ;
static uint32_t gReceivedFrameCount = 0 ;
static uint32_t gSentFrameCount = 0 ;

bool Send_CANFD_Frame (CANFDMessage &msg, int ID , int len ,unsigned char b0,unsigned char b1,unsigned char b2,unsigned char b3,unsigned char b4,unsigned char b5,unsigned char b6,unsigned char b7 )

{
msg.id=ID;
msg.len=len;
msg.idx=255;
msg.data [0] = b0 ; // First data byte
msg.data [1] = b1 ; // First data byte
msg.data [2] = b2 ; // First data byte
msg.data [3] = b3 ; // First data byte
msg.data [4] = b4 ; // First data byte
msg.data [5] = b5 ; // First data byte
msg.data [6] = b6 ; // First data byte
msg.data [7] = b7 ; // First data byte

return can.tryToSend (msg) ;
}

 void loop ()
{
  can.poll () ;
  CANFDMessage msg ;
/*
 if (gSendDate < millis ())
  {
    gSendDate += 2000;
    const bool ok = Send_CANFD_Frame (msg, 0x123 , 8 , 0x01 , 0x12 , 0xCF, 0x15 , 0xF1, 0x56, 0x56, 0x10 ) ;
    
    if (ok) 
    {
      gSentFrameCount += 1 ;
      Serial.print ("Sent: ") ;
      Serial.println (gSentFrameCount) ;
    } 
    else 
    {
      Serial.println ("Send failure") ; 
    } 
  } */

----------------RECEIVE-------------

   if(can.receive (msg)) 
    {
      if (msg.rtr && msg.ext && (msg.id == 0x123)) 
      {
       Serial.println("message>");
       handle_myMessage_0 (msg) ; // Standard data frame , id is 0x234
       
      }
      
    }
}

void handle_myMessage_0 (const CANFDMessage & inMessage)
{
    gReceivedFrameCount ++ ;
    Serial.print ("Received: ") ;
    Serial.println (gReceivedFrameCount) ;
}
 
//——————————————————————————————————————————————————————————————————————————————