pierremolinaro/acan2517FD #2: SPI doesn't work on Arduino UNO

zetora, 2019-02-06:
Thanks a lot Mr.Molinaro,

  I'm trying to use 2517FD with an ATMega328P (arduino UNO), but it seems to SPI doesn´t work.  Could you help me please?

Many thanks!
                           Jan
  
Arduino 1.8.8
I'm using standard SPI.h  Arduino library.

Communication on serial monitor:

Hello
Error Can: 0x4
Sent: 1

No communication on MOSI/MISO/SCK ! (measured by oscilloscope).
By using direct SPI communication only (without your  ACAN2517FD.h driver)  SPI is working (measured by oscilloscope).

PINs:  MOSI  - PCINT3  (11)
          MISO  - PCINT4  (12)
          SCK     - PCINT5  (13)
          CS       - PCINT2  (10)
          INT      - PCINT19 (3)

Standard SPI PINs  -> I'm using   SPI.begin () ;  

Used 20MHz crystal.

CODE: ( from your examples - only for HW function testing) 

#include <ACAN2517FD.h>
#include <SPI.h>

static const byte MCP2517_CS  = 10 ; // CS input of MCP2517
static const byte MCP2517_INT = 3 ; // INT output of MCP2517

ACAN2517FD can (MCP2517_CS, SPI, MCP2517_INT) ;

void setup () {
  pinMode(A0, OUTPUT);  //LED1
  pinMode(A1, OUTPUT);  //LED2
  pinMode(A5, OUTPUT);  //PC5  
  digitalWrite (A5, LOW) ; //PC5  -  CAN_STB transceiver (Normal mode)      

    Serial.begin (9600) ;
    SPI.begin () ;    
    while (!Serial) {}
    Serial.println ("Hello") ;
    // Arbitration bit rate: 125 kbit/s, data bit rate: 500 kbit/s
    ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_20MHz, 125 * 1000, ACAN2517FDSettings::DATA_BITRATE_x4) ;
    settings.mRequestedMode = ACAN2517FDSettings::InternalLoopBack ; // Select loopback mode
    const uint32_t errorCode = can.begin (settings, [] { can.isr () ; }) ;
    if (0 == errorCode) {
      Serial.println ("Can ok") ;
    } else {
      Serial.print ("Error Can: 0x") ;
      Serial.println (errorCode, HEX) ;
    }
  }

static unsigned gSendDate = 0 ;
static unsigned gSentCount = 0 ;
static unsigned gReceivedCount = 0 ;

void loop () {
  CANFDMessage message ;
  if (gSendDate < millis ()) {
    message.id = 0x542 ;
    const bool ok = can.tryToSend (message) ;
    if (ok) {
      gSendDate += 2000 ;
      gSentCount += 1 ;
      Serial.print ("Sent: ") ;
      Serial.println (gSentCount) ;
    }
  }
  if (can.receive (message)) {
    gReceivedCount += 1 ;
    Serial.print ("Received: ") ;
    Serial.println (gReceivedCount) ;
  }
}

pierremolinaro, 2019-02-06:
Hello,

I have just read your code. The error 0x4 is kTooFarFromDesiredBitRate, meaning that the bit rate cannot be achieved.

> ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_20MHz, 125 * 1000, ACAN2517FDSettings::DATA_BITRATE_x4) ;
But 125 * 1000 is not 125 kbit/s, because in Arduino Uno, int are 16-bits, so 125 * 1000 is actually (125 * 1000) modulo 65536 = 59 464  So the actual bit rate 59 464 bit/s. And the bit rate computing function cannot find a valid setting for this value.

Call settings.actualArbitrationBitRate() for getting the actual bit rate.

In arduino Uno, you should write the bit rate as 125UL * 1000UL, enforcing the values are 32-bits.

Best regards,

Pierre Molinaro

zetora, 2019-02-07:
Hello,

    Thank you for your reply.  But,  I was trying to use  ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_20MHz, 500ul * 1000ul , ACAN2517FDSettings::DATA_BITRATE_x4) ; after your reply (In arduino Uno, you should write the bit rate as 125UL * 1000UL, enforcing the values are 32-bits.)

for 20MHz Crystal and 500kbit/s arbitration  and  2Mbit/s data rate.   Supported data rate settings for 20MHz crystal:   „500 kbit/s   1x 2x 4x 5x 8x“

Result:

Hello
Error Can: 0x4
mArbitrationBitRateClosedToDesiredRate: 0
Actual Arbitration Bit Rate: 0
Actual Data Bit Rate: 0
distance: 999996
Bit rate prescaler: 256
Arbitration Phase segment 1: 153
Arbitration Phase segment 2: 38
Arbitration Resynchronization Jump Width: 38
Arbitration Sample Point: 80
Data Phase segment 1: 32
Data Phase segment 2: 15
Data Resynchronization Jump Width: 15
Data Sample Point: 68
Consistency: 0
I really don´t know what’s wrong on Adruino UNO   -   Actual Arbitration Bit Rate: 0

I need to use this driver for 500kbit/s arbitration and  2Mbit/s data rate only by CAN FD mode.

In other projects to use it on 500kbit/s in normal Hi-speed CAN mode.  Do you have any Idea to set it up for ATmega238P?

Many thanks Pierre!

Greetings from the Czech Republic.

                                                       Jan

pierremolinaro, 2019-02-07:
Hello,

I have got an Arduino Uno and a MCP2517FD Click board from Mikroe. Previously, I did not test the library on Arduino Uno, only on Teensy 3.x, Adafruit Feather M0, and recently on ESP32.

In fact, the current release of ACAN2517FD library (1.1.2) does not work on Arduino Uno.

I have made the bug fixes and will provide this evening or tomorrow a new release (1.1.3) with a demo sketch LoopBackDemoArduinoUno.

Note the following points:
	- it is very very important to put a resistor (10 kΩ) between CS and VDD of MCP2517FD; without this resistor, CS is floating from power-on until CS is configured as OUPUT; it seems that may crash the MCP2517FD ;
	- an Arduino Uno has only 2048 bytes, so you should override default values for driver buffer sizes:
  		settings.mDriverTransmitFIFOSize = 1 ;
  		settings.mDriverReceiveFIFOSize = 1 ;

Un grand bonjour de France !

Pierre

zetora, 2019-02-08:
Hello Pierre,

     ☺ It works ☺   Version 1.1.3 works with Arduino UNO on 2Mbit/s  CAN FD data rate.
I will test your ACAN2517 (CAN 2.0B) Library on A-Uno on Monday next week too.
Many, many thanks!
                            Jan
From: Pierre Molinaro <notifications@github.com>
Sent: Thursday, February 7, 2019 7:58 PM
To: pierremolinaro/acan2517FD <acan2517FD@noreply.github.com>
Cc: Zdenek, Jan (EPE) <Jan.Zdenek@skoda-auto.cz>; Author <author@noreply.github.com>
Subject: [SA-Marketing Mail] Re: [pierremolinaro/acan2517FD] SPI doesn't work on Arduino UNO (#2)

Hello,

I have got an Arduino Uno and a MCP2517FD Click board from Mikroe. Previously, I did not test the library on Arduino Uno, only on Teensy 3.x, Adafruit Feather M0, and recently on ESP32.

In fact, the current release of ACAN2517FD library (1.1.2) does not work on Arduino Uno.

I have made the bug fixes and will provide this evening or tomorrow a new release (1.1.3) with a demo sketch LoopBackDemoArduinoUno.

Note the following points:
- it is very very important to put a resistor (10 kΩ) between CS and VDD of MCP2517FD; without this resistor, CS is floating from power-on until CS is configured as OUPUT; it seems that may crash the MCP2517FD ;
- an Arduino Uno has only 2048 bytes, so you should override default values for driver buffer sizes:
settings.mDriverTransmitFIFOSize = 1 ;
settings.mDriverReceiveFIFOSize = 1 ;

Un grand bonjour de France !

Pierre

pierremolinaro, 2019-02-08:
Hello Jan,

I am glad it is working.

I had not fully tested the library (Teensy 3.x, ESP32, Adafruit Feather M0), so I had not yet published it. I see that it works on Arduino Uno, so I publish it now. The problem of ACAN2517 1.1.2 is the same as ACAN2517FD 1.1.2, so I think the ACAN2517 1.1.3 wil run on Arduino Uno.

Best regards,

Pierre

zetora, 2019-02-11:
Hello Pierre,

I tested  64byte data message on CAN FD, but data are in first 8 bytes only.   ☹

[cid:image002.png@01D4C1E4.9C10ADB0]

//--- Make an even identifier for MCP2517
    gMessageSentByMCP2517.id = 0x22;
    gMessageSentByMCP2517.len = 64 ;
    gMessageSentByMCP2517.ext = true ;
    for (uint8_t i=0 ; i<64 ; i++) {
      gMessageSentByMCP2517.data [i] = i ;
      Serial.println(gMessageSentByMCP2517.data [i]);
   }

  //--- Send frame via MCP2517
    const bool ok = can.tryToSend (gMessageSentByMCP2517) ;
    if (ok) {
      gSentFrameCount2517 += 1 ;
      gMCP2517ReadyToSend = false ;

Many thanks

                        Jan

pierremolinaro, 2019-02-11:
Hello,

What is the type of gMessageSentByMCP2517 ? It should be CANFDMessage. If it is CANMessage, it is limited to 8.

Sincerly yours,

Pierre

zetora, 2019-02-12:
Hello Pierre,

My mistake. It Works!

static CANMessage gMessageSentByMCP2517 ;     >>>>>>  static CANFDMessage gMessageSentByMCP2517 ;

Many Thanks

                        Jan

From: Pierre Molinaro <notifications@github.com>
Sent: Monday, February 11, 2019 7:33 PM
To: pierremolinaro/acan2517FD <acan2517FD@noreply.github.com>
Cc: Zdenek, Jan (EPE) <Jan.Zdenek@skoda-auto.cz>; Author <author@noreply.github.com>
Subject: [SA-Marketing Mail] Re: [pierremolinaro/acan2517FD] SPI doesn't work on Arduino UNO (#2)

Hello,

What is the type of gMessageSentByMCP2517 ? It should be CANFDMessage. If it is CANMessage, it is limited to 8.

Sincerly yours,

Pierre

closed 2019-03-29 by pierremolinaro