pierremolinaro/acan2517FD #3: No data rate switching in CAN FD mode

danielhenz, 2019-03-19:
Bonjour M. Molinaro and thank you for your work!

I tried to establish a CAN FD communication, using two Arduino Nano clones and two MCP2517FD click boards from Mikroelektronika, connected with a short DB9 cable (driver version 1.1.3). One is sender only, the other one receiver only.

It works perfectly transmitting some CAN FD Messages with 64 bytes payload, up to 1Mbit/s. But the data rate for the payload is not switched, as I can view on the scope. Even 125kbit/s arbitration rate and a x2 factor for data rate is ignored. 

I found out that the the method ACAN2517FD::appendInControllerTxFIFO() is called, but the BRS bit is not set (as in the method  ), so I copied the lines 

```
if (mHasDataBitRate) { // Copied from sendViaTXQ()
  data |= 1 << 6 ; // Set BRS bit
} 
```

into this method. 

Now, the sending controller seems to increase the data rate for the first bit and fails immediately. This is done exactly 32 times in a fast sequence, until it gives up. There is no error state, I can try to send the frames cyclically and it appears on the bus. Only, there is no valid data transmission. 

Did you try the data rate switch using two MCP2517FD controllers? Am I missing an important configuration detail?

I did only slight modifications to your loopback example code, e.g.

```
ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_40MHz, 125UL * 1000UL, ACAN2517FDSettings::DATA_BITRATE_x4) ;

//settings.mRequestedMode = ACAN2517FDSettings::InternalLoopBack ; // No loopback here!

settings.mDriverTransmitFIFOSize = 2;
settings.mDriverReceiveFIFOSize = 2;
 
```

By the way: 
Shouldn‘t it be 

`d = mUsesTXQ ?  (1 << 4) : 0x00 ; 
`
instead of 

`d = mUsesTXQ ?  0x04 : 0x00 ;`

in ACAN2517FD::begin() ?

Kind regards, 

Daniel

danielhenz, 2019-03-19:
Now the bitrate switch works with my patch to set the BRS bit. 

The measurement influenced the CAN communication, but I fixed it.

pierremolinaro, 2019-03-20:
Sorry for the bug, thank very much for your patch.

Il will check this :

> By the way:
> Shouldn‘t it be
> 
> d = mUsesTXQ ? (1 << 4) : 0x00 ;  
> instead of
> 
> d = mUsesTXQ ? 0x04 : 0x00 ;
> 
> in ACAN2517FD::begin() ?
I plan to provide a new release at this end of this week.

Thank you for your message,

Pierre

danielhenz, 2019-03-22:
I thank you for integrating this!

Kind regards, 

Daniel

closed 2019-03-29 by pierremolinaro