pierremolinaro/acan2517FD #4: Issues initializing ACan2517FD with Arduino Due

tlewiscpp, 2019-04-09:
My code for initialization is:
```
    # define MCP2517_CS 9
    # define MCP2517_INT 2
    canController = new ACAN2517FD{MCP2517_CS, SPI, MCP2517_INT};
     
     // Arbitration bit rate: 125 kbit/s, data bit rate: 500 kbit/s
    ACAN2517FDSettings settings{ACAN2517FDSettings::OSC_40MHz, 125 * 1000, ACAN2517FDSettings::DATA_BITRATE_x4};
    uint32_t errorCode{0};
    for (uint8_t i = 0; i < CAN_INITIALIZATION_RETRY_COUNT; i++) {
        errorCode = canController->begin (settings, [] { canController->isr(); });
        if (errorCode == 0) {
            SerialUSB.println ("Can OK") ;
            return errorCode;
        } else{
            SerialUSB.print ("Error Can: 0x") ;
            SerialUSB.println (errorCode, HEX) ;
            delay(50);
        }
    }
```

I am using the following breakout board: [CAN](https://copperhilltech.com/can-fd-breakout-board-with-spi-interface/) with the following wiring:
Arduino    |    CAN-FD
50 (MOSI) -> DO
51 (MISO) -> DI
52 (SCK) -> SCK
5v          -> 5v
3.3v        -> 3.3v
9 (CS)     ->  CS
2 (INT)    -> INT

And I get the error code 1 (kRequestedConfigurationModeTimeOut). I have the jumper on the board set to 3.3v voltage levels. Can you see anything I can try or something I'm missing? Does CLKO need to be landed somewhere? I also don't know what the oscillator is, but I've tried every one of the ACan2517FDSettings::Oscillator enums and get the same result. I have also tried switching DO/DI. Thanks.

pierremolinaro, 2019-04-09:
Hello,

I suppose that you have a Teensy 3.5 or a Teensy 3.6 and you use SPI2.

Alternate pins for SPI2 are:
	MISO2 -> 51
	MOSI2 -> 52
	SCK2 -> 53
The initialisation code should be :
	canController = new ACAN2517FD{MCP2517_CS, SPI2, MCP2517_INT};
	SPI2.setMOSI (52) ;
	SPI2.setMISO (51) ;
  	SPI2.setSCK (53) ;
	SPI2.begin () ;
	ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_40MHz, 125 * 1000, ACAN2517FDSettings::DATA_BITRATE_x4) ;
	const uint32_t errorCode = canController->begin (settings, [] { canController->isr(); });

Calling canController->begin several times is useless : if it fails the first time, it will always fail.

I did not know the MCP2517FD Breakout Board, but I think the connections are the folowing:
	- Teensy MOSI —> MCP2517FD SDI
	- Teensy MISO —> MCP2517FD SDO
	- Teensy SCK —> MCP2517FD SCK
	- Teensy CS —> MCP2517FD CS
	- Teensy INT —> MCP2517FD INT

MCP2517FD CLKO is an output, you can left it unconnected. MCP2517FD INT0 and INT1 are be configured in several ways, the library does not need a particular configuration to work.
 
5V is required

The Teensy 3.6 is 3.3, so 3.3V is required, and JP5 should be on 1-2.
 
The schematic does not specify the frequency of the oscillator, you should read the Y2 value on your board. Microchip recommends a 4 MHz, 20 MHz or 40 MHz quartz.

I hope this will help you.

Pierre

closed 2019-06-02 by pierremolinaro

tlewiscpp, 2019-06-13:
Just FYI for anyone stumbling upon this: The Arduino DUE DOES NOT break out the SPI bus on pins 50-53 like the Arduino Mega 2560 does. You need to use the actual SPI header on the DUE.

pierremolinaro, 2019-06-15:
Hello,

I agree, I have read SPI related pages for Arduino Due, my library does not handle specific SPI settings of this platform.

Best regards,

Pierre Molinaro