pierremolinaro/acan2517FD #6: Operation Mode is not changing using arduino uno Loopback example

gab7891, 2019-07-04:
Hi, many thanks for this library,
I am using an Arduino uno and the LoopBackDemo example and when the REQOP is changed to 010 (Set Internal Loopback mode) nothing happens and the deadline is reached. I checked the MODIF flag and it keeps in 0 and the OPMOD always keeps in 100 (Configuration mode).
The SPI communication is working  and I Set the MODIE bit to enable Mode change Interruptions.
Is the arduino uno the one that should take care of the interruption or when the REQOP is changed the MCP2517FD controller should do it by it self?
Do you know what could it be for the OPMOD don't change?
Thanks.
Best Regards,
Gabriel Pinheiro.

pierremolinaro, 2019-07-04:
Hi, 

What is the error code value returned by begin method ?

Best regards,

Pierre

gab7891, 2019-07-04:
Hi, thanks for the answer,
The error code is 0x10000 and it is in this part of the code that fails to
change the operation mode:

 writeByteRegister (C1CON_REGISTER + 3, inSettings.mRequestedMode);
  //----------------------------------- Wait (2 ms max) until requested
mode is reached
    bool wait = true ;
    const uint32_t deadline = millis () + 2 ;
    while (wait) {
      const uint8_t actualMode = (readByteRegister (C1CON_REGISTER + 2) >>
5) & 0x07 ;
      wait = actualMode != inSettings.mRequestedMode ;
      if (wait && (millis () >= deadline)) {
        errorCode |= kRequestedModeTimeOut ;
        wait = false ;
      }
    }
Best regards,
Gabriel Pinheiro.

pierremolinaro, 2019-07-06:
Hello,

The error code says that the MCP2517FD cannot reach the requested mode within 2 ms.
	- Check the MCP2517FD connections (SCK, MISO, MOSI, CS, INT) ;
	- What is the frequency of the quartz attached to the MCP2517FD, have you selected the parameter of begin method accordingly ?

Note the MCP2717FD has no RESRT pin, so sometimes when things go wrong, it cannot be reseted. Power off and then power on solves the issue in such case.

Best regards,

Pierre

gab7891, 2019-07-08:
Hi,
So, I checked the connections and they are all working. I am using the breakout board CAN FD with SPI interface from SKPANG. It uses a 40 MHz oscillator and I selected the parameter accordingly and still it does not work. When the REQOP is set to the desire operation mode nothing happens and the OPMOD always keep at 0b100, which is in operation mode...
I saw that it says this in the beginning of the library: " Very very important: put a 10kΩ resistor between CS and VDD of MCP2517FD". I am not using the resistor but I'm not seeing where this can be the solution to this..
What is the purpose of this Resistor?

Best Regards,
Gabriel Pinheiro.

pierremolinaro, 2019-07-09:
Hi,

On power-up, the CS pin is programmed as input, so it is floating. Floating pin may have the active low level, so the MCP2517FD is selected in a some way.
Later, when "begin" is executed, the CS pin programs it as an output.

The problem is the MCP2517 has no RESET pin, RESET is done by software. But I have found that erratic selection can prevent software reset to success.

Adding a resistor between CS and VDD maintain CS high (no selection) after power-up, until CS is configured as output.

Regards,

Pierre

gab7891, 2019-07-09:
Hi, 
I add the resistor and it keeps making the same thing. It simply don´'t change the operation mode.
I bought the mcp2517FD Click to see if it does the same thing because it's very strange that it does't change the operation mode when I change the REQOP bits...
The MCP2517FD controller can be broken and don't change... I don't know, I will test in the MCP2517FD Click board to see if it does the same thing..
After testing I will reply here to tell if it worked.
By the way, this are the values that it's being used by the code:
->ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_40MHz, 125UL * 1000UL, ACAN2517FDSettings::DATA_BITRATE_x1) ;
->Bit Rate Prescaler: 1
->Arbitration Phase segment 1: 255
->Arbitration Phase segment 2: 64
->Arbitration SJW:64
->Actual Arbitration Bit Rate: 125000 bit/s
->Exact Arbitration Bit Rate ? yes
->Arbitration Sample point: 80%
->Configuration error 0x10000

Best Regards,
Gabriel Pinheiro.

pierremolinaro, 2019-07-09:
Quite strange… 

You use a Microchip MCP2517FD CAN FD Breakout Board from SK Pang, what is your platform? Arduino Uno, ESP32, … ? Do you set the JP5 strap accordingly ?

Pierre

pierremolinaro, 2019-07-09:
Another question : is the JP3 jumper is on ? I think it is required, even for internal / external loopback mode.

Pierre

pierremolinaro, 2019-07-09:
Sorry, I did not see you use an Arduino Uno. So JP5 should be between 2 and 3 (+5V selected).

Best regards,

Pierre

gab7891, 2019-07-09:
Hi,
I don't have the JP3 jumper ON..
I will try it now to see if works!
Thanks.

Best regards,
Gabriel Pinheiro

gab7891, 2019-07-10:
Hi,
So I put the JP3 jumper ON and it didn't work...
When the  MCP2517FD Click board arrive I will test with it and I will tell you If it worked, but with the SKPANG CAN FD breakout Board is not changing the operation mode..

Best regards,
Gabriel Pinheiro.

gab7891, 2019-07-12:
Hi,
I already received the MCP2517fd Click board and I'm not able to put the SPI connection working on it..
Can you tell me what are the pins that are used on this board with this library?
I saw that this board has the stby pin, it is needed to use it?

Best regards,
Gabriel Pinheiro.

gab7891, 2019-07-16:
Hi, I already have connection with the MCP2517FD and it changed to Normal FD mode!
So it was the other board that was broken!
But now I'm trying to test it by sending some CAN FD packets and it is not receiving. Can it be the source that is sending with bad timing and the click board never receives the CAN packet?

Best regards,
Gabriel Pinheiro.

gab7891, 2019-07-17:
Hi, It is working now, really thanks for the help and consideration!
One thing, when reeding your code I believe I found an error in the configuration..
In this part: 
  //----------------------------------- Configure ISO CRC Enable bit
    data8 = 1 << 6 ; // PXEDIS <-- 1
    if (inSettings.mISOCRCEnabled) {
      data8 |= 1 << 5 ; //  Enable ISO CRC in CAN FD Frames bit
    }
    writeByteRegister (IOCON_REGISTER, data8); // DS20005688B, page 24 -> C1CON_REGISTER
but it should be writing to the C1CON_REGISTER instead like this:
writeByteRegister (C1CON_REGISTER, data8); // DS20005688B, page 24

Best Regards,
Gabriel Pinheiro.

closed 2019-07-22 by gab7891

lalitheranti, 2019-12-11:
which library are you using for the breakout board ? I need help

gab7891, 2019-12-12:
I am using the acan2517fd library.

Best Regards,
Gabriel Pinheiro

lalitheranti, 2020-02-25:
Hi, Which FD lib are you using currently now? is it the updated one on DEC 31, 2019 or the old commit version? Can you share the lib and sample code or example with CAN FD working on skpang board. I am using the same. would be helpful for my student project.

lalitheranti, 2020-02-25:
Hi, Which FD lib are you using currently? The new one updated on DEc 31, 2019 or older commit version? Could you share the CAN FD lib and sample working example on SK pang board as I am using the same for my student project. It would be much helpful.

gab7891, 2020-02-25:
Hi, I was using the old version until 08/19, but I stoped using the SK pang
board because after many trials and tests I could not put it working as It
should be. Instead I bought the Mikroeletronika CANFD board and It worked
immediatly.
So in my case, the problem was the board, not the code..

Best Regards,
Gabriel Pinheiro.

lalitheranti, 2020-02-25:
Interesting to note. I have the Mikroeletronika CANFD board as well MCP2518FD click board. but still the same issue as 
[image]

[image]

lalitheranti, 2020-02-25:
It started working now when I change the // false --> Do NOT include Stuff Bit Count in CRC Field and use CRC Initialization Vector with all zeros
// true --> Include Stuff Bit Count in CRC Field and use Non-Zero CRC Initialization Vector according to ISO 11898-1:2015
  public: bool mISOCRCEnabled = false ;

from true. :)

pierremolinaro, 2020-02-25:
What is the value returned by can.begin ?

You can try to use less RAM by adding theses settings before calling can.begin :
  settings.mDriverTransmitFIFOSize = 1 ;
  settings.mDriverReceiveFIFOSize = 1 ;

pierremolinaro, 2020-02-25:
> Hi,
>
> Interesting, what is the reference of this SK pang board ? On the SK Pang web pages, I have only found the ISO-CAN-FD-BRK board, but it contains a transceiver, but no MCP2518FD.
>
> Best regards,
>
> Pierre

Sorry, I find the board on SK PANG, it is MCP2517FD-BRK.