pierremolinaro/acan2517FD #7: Can't loopback with new library with an arduino mega

E-nodev, 2019-09-23:
Hello, 
I tried the sample test LoopBackDemoArduinoUno, before the new release it's ok,
But now i've a Configuration error 0x1 and i can't received anything.
Send return no error.
Can you help me ?

Thank you

pierremolinaro, 2019-09-23:
Hello,

Error 0x1 (kRequestedConfigurationModeTimeOut) means that MCP2517FD cannot enter in configuration mode in 2 ms.

I have noticed that this error sometimes comes after a new sketch is uploaded. I have solved by power off and then power the Arduino board and the MCP2517FD.

I just had the loopbackdemo sketch executed on my teensy, it works without any problems.

Does this error occur always ?

Pierre

E-nodev, 2019-09-24:
Hello, 
The error occur always after 5 try to unplug the board and plug. 
It's ok with your previours library, but with the new i have the same error code.
 
Thank you for your help and your reactivity.

Fred

tizianotosco, 2019-12-03:
Hello,
I tried the LoopBackDemoArduinoUno sample test with ArduinoMega but it doesn't work. Are there any particular configs to modify?
Can you help me?
Thank you

E-nodev, 2019-12-04:
It works with a old version :
https://www.arduinolibraries.info/libraries/acan2517-fd
I don't remember which version sorry :/

Good luck

mpfarr, 2019-12-10:
Hi,
i just started playing with acan2517-fd on Teensy 3.6.

I face an undocumented error:

Using pin #0 for MOSI: yes
Using pin #1 for MISO: yes
Using pin #32 for SCK: yes
Configure ACAN2517FD
MCP2517FD RAM Usage: 2016 bytes
OSC 61
Configuration error 0x20000
Send failure
Send failure

pierremolinaro, 2019-12-10:
Hello,

The 0x20000 error is documented (see table 7 page 39), it is kX10PLLNotReadyWithin1MS. What is the frequency  of the quartz that drives the MCP2517FD ? For using ACAN2517FDSettings::OSC_4MHz10xPLL as parameter of the begin method, you should have a 4 MHz quartz.

Regards,

Pierre Molinaro

mpfarr, 2020-02-14:
Hi,
i tried the LoopBackDemoArduinoUno on Adruino Uno with MCP2518FD click. I always get configuration  error 0x1. I already double checked MOSI/MOSI/SCK/SS.

Any suggestion with Arduino uno?

br,
Matthias

mpfarr, 2020-02-14:
problem solved: logic level of MCP2518FD click board was wrong.

sorry

closed 2021-12-11 by pierremolinaro