pierremolinaro/acan2517FD #10: Error sending a message over CAN

lalitheranti, 2019-12-17:
Hi I am using examples of the project and using MCP2517fd breakout board with Adafruit itsybitsy M4 express to generate CAN messages. But I am getting the following output after changing certain thinngs in example code:

Bit Rate prescaler: 1
Arbitration Phase segment 1: 31
Arbitration Phase segment 2: 8
Arbitration SJW:8
Actual Arbitration Bit Rate: 1000000 bit/s
Exact Arbitration Bit Rate ? yes
Arbitration Sample point: 80%
Data Phase segment 1: 7
Data Phase segment 2: 2
Data SJW:2
TDCO:7
Even, error code 0x1
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0
Sent: 0, received: 0, errors 0, op mode 0, rcved buffer 0, overflows 0

Please help.
what else and what would be wrong possibly

lalitheranti, 2019-12-18:
Can you please create a simple example to send a data over can bus

Flole998, 2019-12-25:
There are already many examples provided, aswell as a great documentation (which is absolutely great, it's super helpful and something many other libraries don't provide).

pierremolinaro, 2019-12-31:
If you want me to help you, you have to tell me precisely: the microcontroller card you are using, the MCP2517FD or MCP2518FD shield, the ports used for the connections (SCK, MOSI, MISO, CS and INT), and how the "internalLoopBack" sketch has been adapted for your configuration.

An important point is to know the clock frequency of the MCP2517FD, in order to adapt the sketch accordingly.

As Flole998 said, there are many examples in the library.

Regards,

Pierre Molinaro

lalitheranti, 2020-01-13:
Hi, 

Here is my code:

#include "SparkFunLSM6DS3.h"
#include "SPI.h"
#include <ACAN2517.h>

int MCP2517_CS  = 7 ;
int LSM_CS  = 13 ;
LSM6DS3 Sensor(SPI_MODE, LSM_CS);

char buffer1[14];
char buff1[6];
char buff2[2];
char buff3[6];
char addr[14]; 
int16_t data1[3];
int16_t data2[1];
int16_t data3[3];
float ax,ay,az,gx,gy,gz,t;
double afact = 0.000061036088;
double gfact = 0.030518043793;
double toffset = 0;

ACAN2517 can (MCP2517_CS, SPI, 255) ;

void setup() {
 
   pinMode(LSM_CS, OUTPUT);
   pinMode(MCP2517_CS, OUTPUT);
   addr[0] = 0x20; addr[1] = 0x21; addr[2] = 0x22; addr[3] = 0x23; addr[4] = 0x24; addr[5] = 0x25; addr[6] = 0x26; addr[7] = 0x27; addr[8] = 0x28; addr[9] = 0x29; addr[10] = 0x2A; addr[11] = 0x2B; addr[12] = 0x2C; addr[13] = 0x2D;
 Serial.begin(115200);
  if( Sensor.begin() != 0 )
  {
    Serial.println("Problem starting the sensor with CS @ Pin 9.");
  }
  else
  {
    Serial.println("Sensor with CS @ Pin 9 started.");
  }
  ACAN2517Settings settings (ACAN2517Settings::OSC_40MHz, 1000 * 1000) ; 
  settings.mRequestedMode = ACAN2517Settings::Normal20B ; 
  const uint32_t errorCode = can.begin (settings, NULL) ;
 // SPI.begin() ;

}

void loop() {

    SPI.begin() ;
    digitalWrite(LSM_CS, LOW);
    SPI.transfer(0x88|0x0F);
    int whoami = SPI.transfer(0x00);
//  Serial.println(whoami, HEX);
    digitalWrite(LSM_CS, HIGH);
    SPI.end(); 
    readsensordata();
    sendcandata();
    //delay(10);  
              
}

void readsensordata(){
        
        for(int k=0; k<=13; k++){
           SPI.begin() ;
           digitalWrite(LSM_CS, LOW);
           SPI.transfer(addr[k] | 0x80);
           buffer1[k]=SPI.transfer(0x00);
           digitalWrite(LSM_CS, HIGH);
           SPI.end();
           //Serial.println(buffer1[k], HEX);
        }
      
   buff2[0]=buffer1[0];buff2[1]=buffer1[1];
   buff3[0]=buffer1[2];buff3[1]=buffer1[3];buff3[2]=buffer1[4];buff3[3]=buffer1[5];buff3[4]=buffer1[6];buff3[5]=buffer1[7];
   buff1[0]=buffer1[8];buff1[1]=buffer1[9];buff1[2]=buffer1[10];buff1[3]=buffer1[11];buff1[4]=buffer1[12];buff1[5]=buffer1[13];

   data1[0] = buffer1[8] | buffer1[9]<<8;
   data1[1] = buffer1[10] | buffer1[11]<<8 ;
   data1[2] = buffer1[12] | buffer1[13]<<8; 
 
   data2[0] =  buffer1[0] | buffer1[1]<<8;

   data3[0] =  buffer1[2] | buffer1[3]<<8;
   data3[1] =  buffer1[4] | buffer1[5]<<8;
   data3[2] =  buffer1[6] | buffer1[7]<<8;

   ax = afact*data1[0];
   ay = afact*data1[1];
   az = afact*data1[2];

   t = ((data2[0])/256)+25-toffset ;

   gx = gfact*data3[0];
   gy = gfact*data3[1];
   gz = gfact*data3[2];

  // sprintf("%+1.4f,%+1.4f,%+1.4f,%+3.1f,%+3.1f,%+3.1f,%+2.1f\r\n", ax,ay,az,gx,gy,gz,t);
   Serial.print(ax);Serial.print(",");Serial.print(ay);Serial.print(",");Serial.print(az);Serial.print(",");Serial.print(gx);Serial.print(",");Serial.print(gy);Serial.print(",");Serial.print(gz);Serial.print(",");Serial.println(t);
  
//   Serial.print(buff3[4], HEX);
//   Serial.println(buff3[5], HEX);

}

void sendcandata(){
  
  can.poll();
  CANMessage temperature; CANMessage acceleration; CANMessage gyro;
         temperature.id = 0x5A6;
         temperature.len = 2;
         temperature.data[0] = buff2[0] ; temperature.data[1] = buff2[1] ;
         acceleration.id = 0x5A4;
         acceleration.len = 6;
         acceleration.data[0] = buff1[0];   acceleration.data[1] = buff1[1]; acceleration.data[2] = buff1[2]; acceleration.data[3] = buff1[3]; acceleration.data[4] = buff1[4]; acceleration.data[5] = buff1[5];
         gyro.id = 0x5A5;
         gyro.len = 6;
         gyro.data[0] = buff3[0]; gyro.data[1] = buff3[1]; gyro.data[2] = buff3[2]; gyro.data[3] = buff3[3]; gyro.data[4] = buff3[4]; gyro.data[5] = buff3[5];
   SPI.begin() ;
   digitalWrite(MCP2517_CS, LOW);
         can.tryToSend(acceleration);
         can.tryToSend(gyro);
         can.tryToSend(temperature);
   digitalWrite(MCP2517_CS, HIGH);  
   SPI.end();
   
}

I am using the MCP2517 device breakoutboard from skpang: http://skpang.co.uk/catalog/microchip-mcp2517fd-can-fd-breakout-board-p-1544.html

I am able to send data over can, but the problem is that sometimes I dont see output in the PCAN explorer.Only sometimes I see. Why is it so? 

what should I change?

Flole998, 2020-01-13:
Just to be sure: You are using acan2517 and not acan2517fd? You filed this issue in the acan2517fd repo so I just want to make sure.

lalitheranti, 2020-01-13:
I am using currently ACAN2517 lib to send my sensor data over can bus as normal CAN 2.0 messages. But it goes similar with ACAN2517fd lib, that I can see the data only sometimes and not consistently. like if i change something inthe code and compile, i dont see it immediately. I have to wait sometime and unplug everything, do it multile times sometimes and then I see the output in PCAN explorer. It is strange that I dont understand why it is like that. Is it the problem with the code or the device? i dont get it

Flole998, 2020-01-13:
In the docs there are several problems mentioned with the MCP2517, I am using MCP2518FD without any issues. Maybe it's worth trying to change the Controller to the 2518FD. You should also not do the SPI.begin and SPI.end when sending a CAN Message and should not touch CS of the MCP251{7|8}[FD] at all and let the library handle that. Please also check if there is really a 40Mhz Oscillator on your PCB.

lalitheranti, 2020-01-13:
Thanks for suggestions. I have MCP2518FD click board as well, I tried it with your suggestions. Still same issue. there is 40MHz on it. I dont know why. is it because there are 2 spi devices and a long cable arround 1m for sensor spi data?

Flole998, 2020-01-13:
1m is a lot for 10Mhz SPI communication. I suggest lowering the SPI speed if you use that long cables. SPI was designed for communication on the same PCB, not for that long wires.

lalitheranti, 2020-01-14:
I used resistors arround 50 ohm on each line for the sensor spi cable and I can see the sensor data output on the serial monitor of arduino but, i have normal short bread board cables for the CAN controller to send the data over same spi on microcontroller. Should I also use resistors for the can spi? I will check with lowering spi speed as well.

lalitheranti, 2020-01-14:
I tried lowering spi speed to 500Khz but still I cannot see the CAN data :(

Flole998, 2020-01-16:
Could you check if errorCode is != 0 at the beginning? If it is 0 I suggest using a Logic Analyzer to figure out what is being sent and received. That's at least how I would approach this problem.

closed 2020-03-17 by lalitheranti