pierremolinaro/acan2517FD #12: 10K ohm resistor on CS pin

lalitheranti, 2020-02-19:
Hi, where should I connect the 10k ohm resistor? I see in the manual that between vcc and CS line, but I couldnt find exactly what vcc pin is. is it 3.3v pin or 5v pin on mcp2517fd?

pierremolinaro, 2020-02-19:
Vcc pin is VDD pin (#14) of the MCP2517FD (or MPC2518FD).

If MCP2517FD is powered with +5V, use +5V. If it is powered with +3.3V, use +3.3V.

Best Regards,

Pierre

lalitheranti, 2020-02-19:
Ok thanks. but I am using both 5v and 3.3v supplied to the breakout board. : http://skpang.co.uk/catalog/microchip-mcp2517fd-can-fd-breakout-board-p-1544.html.

probably then i Use 5V to connect 10k

pierremolinaro, 2020-02-19:
Yes, but what is your microcontroller ? Following the schematics, for example :
	- Arduino Uno : 5V microcontroller -> JP5 on 2-3
	- Teensy 3.2 : 3.3 microcontroller -> JP5 on 1-2

And connect the 10kΩ between MCP2517FD pin 14 (VDD) and MCP2517FD pin 13 (nCS).

Pierre

lalitheranti, 2020-02-19:
I am using Adafruit itsybitsy M4 express

pierremolinaro, 2020-02-19:
It is a 3.3V microcontroller: all logic is 3.3V. See "Logic Pins" at https://learn.adafruit.com/introducing-adafruit-itsybitsy-m4/pinouts <https://learn.adafruit.com/introducing-adafruit-itsybitsy-m4/pinouts>

lalitheranti, 2020-02-19:
I am having issues with CANFD working. I used the libs for my project and normal CAN is working from ACAN2517 Lib and for FD I am using ACAN2517FD, but it is not working. I am using PCAN to view. it shows following error. 
[image]

lalitheranti, 2020-02-19:
> It is a 3.3V microcontroller: all logic is 3.3V. See "Logic Pins" at https://learn.adafruit.com/introducing-adafruit-itsybitsy-m4/pinouts <https://learn.adafruit.com/introducing-adafruit-itsybitsy-m4/pinouts>
> […](#)

ofcourse, I am using 3.3V logic only. But I am powering the CANFD breakout boards 5V pin via USB output pin of itsybitsy as the breakout needs both 5v and 3.3v.

pierremolinaro, 2020-02-19:
I did not know PCAN-Explorer, but I have git it after a search in the Web.

What is your PCAN-Explorer release ? Release 5 works only for CAN, so it always raises an error on a CANFD frame.  Release 6 works on CANFD and CAN frame. I think you have release 6 because the Bit Rate field is "Nominal 1Mbit/s, Data 4Mbit/s".

Note also there are two incompatible CANFD formats : ISO CANFD, and non ISO CANFD. ACAN2517FD uses  ISO CANFD by default. What is the setting of PCAN-Explorer 6 ?

pierremolinaro, 2020-02-19:
> … <x-msg://5/#>

+5V is required for MCP2562FD.  For Adafruit itsybitsy M4 express, 3.3V is required for MCP2517FD, so put JP5 between 1-2, and the 10kΩ between 3.3V and CS.

lalitheranti, 2020-02-19:
I think it is ISO standard only by default., although i am not able to see the setting. Before the update of December 31, 2019. I had a working FD comunication as shown in the attached screen shots. But now, I am using the new LIb updated one, i dont see it is working and I even tried with the old lib, it is not working. I am not sure what I am doing wrong.
[image]
[image]

lalitheranti, 2020-02-24:
I get SPI connection right, but the trytosend method returns false for FD messages with 2518FD click board.

lalitheranti, 2020-02-25:
It started working now when I change the // false --> Do NOT include Stuff Bit Count in CRC Field and use CRC Initialization Vector with all zeros
// true --> Include Stuff Bit Count in CRC Field and use Non-Zero CRC Initialization Vector according to ISO 11898-1:2015
  public: bool mISOCRCEnabled = false ;

from true. :)

closed 2020-03-17 by lalitheranti