pierremolinaro/acan2517FD #15: Possible Bug regarding Retransmission Setting

Flole998, 2020-05-28:
The datasheet of the MCP2518FD says on page 27:

> RTXAT: Restrict Retransmission Attempts bit( 1)
> 1 = Restricted retransmission attempts, CiFIFOCONm.TXAT is used
> 0 = Unlimited number of retransmission attempts, CiFIFOCONm.TXAT will be ignored

Current code says:
`data8 = mUsesTXQ ? (1 << 4) : 0x00 ; // Bug fix in 1.1.4 (thanks to danielhenz)
`
I changed it to

```
data8 = 0x01 ; //Enable RTXAT to limit retransmissions (Flole)
data8 |= mUsesTXQ ? (1 << 4) : 0x00 ; // Bug fix in 1.1.4 (thanks to danielhenz)
```

Do I see that correctly that right now RTXAT is not set, causing the retransmissions to be always unlimited and the code should be modified just as I did it above?

pierremolinaro, 2020-05-30:
Hello,

Yes, your code sets the RTXAT bit. This enables the CiFIFOCONm.TXAT bits to control retransmission. As these bits remain clear, retransmission attempts are disabled.

Best regards,

Pierre

Flole998, 2020-05-30:
That would mean for this library that we should add my modification, as we set the TXAT Bits based on the users selection. In the current version we are always doing unlimited retransmissions, even though the options for three retransmissions or no retransmissions are set.

pierremolinaro, 2020-05-30:
Ok, I do the modifications, and will publish the release tomorrow or on monday.

Best regards,

Pierre Molinaro

Flole998, 2020-06-02:
Thank you very much!

closed 2020-06-02 by Flole998