pierremolinaro/acan2517FD #16: Problem with SPI Communication

lheusser, 2020-07-04:
Dear all

I'm trying to set up a board using Microchip's MCP2518 controller ([MCP2518FD CLICK Board](https://www.mikroe.com/mcp2518fd-click)) together with an Arduino Uno, but I can't get SPI communication working.

The example in ACAN2517FD\examples\LoopBackDemoArduinoUno\LoopBackDemoArduinoUno.ino seems to not be able to setup the SPI correctly.

Here a diagram of the SPI CLK (CH1) and SPI MOSI (CH2) using the example:

[image]

As check I try the Arduino Barometric Pressure Sensor SPI example and here the SPI seems to be OK:

[image]

Any suggestion how to proceed to find the root cause? Thanks in advance!

Ludwig

pierremolinaro, 2020-07-04:
Hello,

Maybe it's a memory overflow: an Arduino has little memory. Try to reduce the size of the transmit and receive buffers:
	settings.mDriverTransmitFIFOSize = 0 ; // Should be >= 0
	settings.mDriverReceiveFIFOSize = 1 ; //Should be > 0

Otherwise, post your code. Do you try to run one of the sample codes ?

Best Regards,

Pierre

pierremolinaro, 2020-07-04:
And you use an MCP2518FD CLICK Board with an Arduino Uno : do you select +5V for VIO ?

Pierre

soso49, 2020-10-23:
Hello, I've had a hard time setting the SPI com with my arduino UNO and a MCP2518FD click board. I haven't seen a clear wiring scheme yet so I'm sharing mine. 
[image]
 **Make sure to remove the jumper on the 3.3V section and solder it on the 5V section.**

pierremolinaro, 2020-10-30:
Thank you very much, I will add this wiring scheme to the library documentation.

Best Regards,

Pierre

dost0011, 2022-03-22:
Hello, I have the same problem and can't find the solution.
Arduino Mega 2560 Rev. 3 --> MCP2517FD click
I have downloaded your lib and tried the example "LoopBackDemoArduinoUnoNoInt".
I connected the CS to PIN 53 (and changed the code). I have used a Pullup 10kOhm. 

Jumpers:
VIO 5V
STBY ON
40 MHz
STBY SEL STBY

51 MOSI --> 6 SDI
50 MISO --> 5 SD0
52 SCK --> 4 SCK
53 CS --> 3 CS
I checked the connections 1000 times.

The error is always 1 = kRequestedConfigurationModeTimeOut.

It looks, that SPI from Arduino Mega doesn't work. I tried settings.mDriverTransmitFIFOSize = 0 --> no effect.

I don't have any single idea... Could you help me?
Many thanks...

dost0011, 2022-03-22:
One thing: is it normal that the voltage between CANL and CANH is 0V if nothing is connected?

pierremolinaro, 2022-04-04:
Hello,

I have checked your connections, I think it is ok.

Did you change the the clock setting ? ACAN2517FDSettings::OSC_40MHz for a 40 MHz clock ?

Pierre

pierremolinaro, 2022-04-04:
> Okay. now I'm one step further with a oscilloskop.
> CANL and CANH are 2.5V --> that's why the difference is 0v
> CLK, MOSI, CS are working. SDO is 0V the whole time.
> Does anybody know, how to wake up the SDO from the MCP2517FD?

I agree, CANH = CANL = 2.5V when the bus is IDLE.

Note the easy way for enabling the ATA6563 transceiver is to connect pins 1 and 2 of strap STBY.

Pierre