pierremolinaro/acan2517FD #19: Setting configuration error 0x10000 with ESP32 and MCP2518FD Click

turbotom93, 2020-10-16:
I'm trying to add CAN FD to my ESP32 project with a Mikroe MCP2518FD CLICK. I am running the LoopBackDemoESP32 example. My only changes were the pin numbers and the oscillator frequency. 

**#define LED_BUILTIN 17**

**static const byte MCP2517_SCK  = 18 ; // SCK input of MCP2517FD**
**static const byte MCP2517_MOSI = 19 ; // SDI input of MCP2517FD**
**static const byte MCP2517_MISO = 23 ; // SDO output of MCP2517FD**

**static const byte MCP2517_CS  = 5 ; // CS input of MCP2517FD**
**static const byte MCP2517_INT = 14 ; // INT output of MCP2517FD**

and 

  **ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_40MHz, 125 * 1000, DataBitRateFactor::x1) ;**

The MCP2518FD is connected like this:
[image]

When I go the terminal I get a config error 0x10000 followed by a bunch of Sent packets, but none Received:
18:16:55.690 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:16:55.690 -> configsip: 0, SPIWP:0xee
18:16:55.690 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:16:55.690 -> mode:DIO, clock div:1
18:16:55.690 -> load:0x3fff0018,len:4
18:16:55.690 -> load:0x3fff001c,len:1216
18:16:55.690 -> ho 0 tail 12 room 4
18:16:55.690 -> load:0x40078000,len:9720
18:16:55.690 -> ho 0 tail 12 room 4
18:16:55.690 -> load:0x40080400,len:6352
18:16:55.690 -> entry 0x400806b8
18:16:55.861 -> sizeof (ACAN2517FDSettings): 44 bytes
18:16:55.861 -> Configure ACAN2517FD
18:16:55.861 -> MCP2517FD RAM Usage: 2016 bytes
18:16:55.861 -> Configuration error 0x10000
18:16:55.861 -> Sent: 1
18:16:57.794 -> Sent: 2
18:16:59.797 -> Sent: 3
18:17:01.807 -> Sent: 4
18:17:03.779 -> Sent: 5
18:17:05.837 -> Sent: 6

pierremolinaro, 2020-10-17:
Hello,

The 0x10000 error code mans that the MCP2518FD cannot reach the requested mode within 2 ms.

The problem is the MCP2518FD has no RESET pin, RESET is done by software. On Power ON,  MCP2517_CS may float, providing erratic selection. I have found that erratic selection can prevent software reset to success.

ESP32 is very tricky at start up, it requires some level on some pins, ans outputs some signal on some pins.

See https://randomnerdtutorials.com/esp32-pinout-reference-gpios/ <https://randomnerdtutorials.com/esp32-pinout-reference-gpios/>, it indicates that IO5 outputs PWM signal at boot. So there are erratic selections at startup, that can prevent MCP2518FD reset to success.

So I suggest to change MCP2517_CS to an other pin, as IO4. And adding a 10kΩ resistor between MCP2517_CS and 3.3V maintain CS high (no selection) after power-up, until MCP2517_CS is configured as output.

Best regards,

Pierre Molinaro

turbotom93, 2020-10-17:
Hi,

Thank you for getting back with me. I made the following modifications:

in the code
**static const byte MCP2517_CS  = 4; // CS input of MCP2517FD**

and in the circuit
[image]

but I am getting the same result:
09:16:39.740 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
09:16:39.740 -> configsip: 0, SPIWP:0xee
09:16:39.740 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
09:16:39.740 -> mode:DIO, clock div:1
09:16:39.740 -> load:0x3fff0018,len:4
09:16:39.740 -> load:0x3fff001c,len:1216
09:16:39.740 -> ho 0 tail 12 room 4
09:16:39.740 -> load:0x40078000,len:9720
09:16:39.740 -> ho 0 tail 12 room 4
09:16:39.740 -> load:0x40080400,len:6352
09:16:39.740 -> entry 0x400806b8
09:16:39.879 -> sizeof (ACAN2517FDSettings): 44 bytes
09:16:39.879 -> Configure ACAN2517FD
09:16:39.879 -> MCP2517FD RAM Usage: 2016 bytes
09:16:39.913 -> Configuration error 0x10000
09:16:39.913 -> Sent: 1
09:16:41.826 -> Sent: 2
09:16:43.828 -> Sent: 3
09:16:45.831 -> Sent: 4
09:16:47.836 -> Sent: 5
09:16:49.840 -> Sent: 6
09:16:51.816 -> Sent: 7

pierremolinaro, 2020-10-19:
Hello,

Sorry, but I cannot explain this. Stupid questions : on the MCP2518FD board, the clock selection is 40MHz, and the VIO SEL is 3V3 ?

Pierre

turbotom93, 2020-10-19:
Hi,

Yes. I am using the board in the default configuration with the 40MHz clock and the 3V3 logic selected. I posted my issue on reddit r/esp32 and a user suggested either inserting a delay before the settings are sent or toggling the /CS line before sending the settings. I tried both but neither produced a different result.  Do you think either of these ideas could work and perhaps I am not implementing them correctly? Is there any way to force the MCP2518FD to reset and try sending the settings again?

turbotom93, 2020-10-20:
I checked the power supplies to the click board and found that I was sourcing 5V from the wrong pin on the ESP32 dev board. I also checked all of the connections between the boards for continuity. One of those two actions managed to fix my problem. The loopback sketch now starts with no errors. I was also able to modify the sendfd-even example and verify I am able to send and receive FD frames! I must sheepishly admit that this one must have been user error. Thank you for your help, the great library, and the thorough documentation.

closed 2020-10-20 by turbotom93

pierremolinaro, 2020-10-20:
Hello,

Unfortunatly, the only way to reset the MCP2518FD is to send a RESET command. There is no known other way, I have checked the MCP2518FD reference manual and read many forums. It's a shame that the MCP2518FD doesn't have a real RESET signal.

Pierre

pierremolinaro, 2020-10-20:
Ok, fine !

Pierre