pierremolinaro/acan2517FD #23: SPI connection with Arduino Nano 33 BLE

Marittia, 2021-07-15:
Hello!
First of all, thanks a lot for the library.

I currently have a MCP2518FD-Click and i wanted to use it along with my Arduino 33 nano BLE.

I already tried the example LoopBackDemoArduinoUno with an Arduino Uno and everything works fine.

But when i switch to the Arduino Nano the code seems to stuck (Arduino IDE stops working as long as i don't disconnect the Arduino from the USB) when calling the function "writeRegister8" inside ACAN2517FD::begin. 
Could it be something wrong with the SPI communication? I double checked the wiring and it seems to be correct (it's equal to the one on my Arduino Uno).

Also, i'm using the MCP2518 with jumper on VIO=3.3 (used a shifter with arduino UNO) and i'm connecting both 3.3V and 5V.

Thanks for your time!

pierremolinaro, 2021-07-17:
Hello!

I have no Arduino 33 nano BLE, so I can't reproduce the bug. But there is no reason why it shouldn't work.

From what I've seen on the Internet, the connection should be :
	D11 - Mosi (SDO)
	D12 - Miso (SDI)
	D13 - SCK
	 CS connection : any pin
	 INT connection : any PIN

It is strange that Arduino IDE stops working as long as i don't disconnect the Arduino from the USB. What pins are you using for CS and INT ? Perhaps there is a conflict with an USB function ? 

Pierre

Marittia, 2021-07-19:
I'm using the same wiring you're describing. 
For CS i'm using pin 6 and for INT pin 7 (i saw that every pin on the Nano is interruptable). I also tried with some other different CS and INT pins, but none of the tries worked out.

Tomorrow i will be able to test it a bit further and see what can cause the problem.

For now, thank you very much!

Marittia, 2021-07-21:
Hello!
Thanks for the help. So far i recognized the "IDE stuck while Arduino is connected" was due to the code blocking (maybe an infinite loop?) after a noInterrupts() call. So what i did was basically commenting all the "noInterrupts()" i found in the library code. The result is that now the Begin function seems to succeed (errorCode = 0).

Still the InternalLoop mode doesn't work, and i saw that in NormalFD mode my can sniffer is able to read only the first packet, all the other ones cannot be sent. Do you happen to have any idea about what could be the problem? 

Thank you very much once again,
Mariano

pierremolinaro, 2021-07-24:
Hello !

It is strange that noInterrupts() call causes code blocking. But the library requires disabling and enabling interrupts, so it's no wonder that by commenting noInterrupts() the library no longer works.

You can try your luck in the following ways (see LoopBackDemoTeensy3xNoInt demo sketch):
	- do not use interrupt pin any more (ast argument of ACAN2517FD object instanciation is 255) ;
	- call can.poll() as often as possible in loop function.

Best Regards,

Pierre

jmFernandezMola, 2022-11-01:
Hi there, 
I have the same problem of Marittia. After some investigations I found this:
_"Note that disabling interrupts on the Arduino boards with native USB capabilities (e.g., Leonardo) will make the board not appear in the Port menu, since this disables its USB capability."_ In the nointerrupts() reference from arduino ([here](https://www.arduino.cc/reference/en/language/functions/interrupts/nointerrupts/)) Arduino Nano 33 BLE seems to be affected by this also. I hope this comment  sheds some light.
Best regards,
JM