pierremolinaro/acan2517FD #25: CANMessage in CANMessage.h

tomtom0707, 2021-09-29:
Hello,
is the following addition possible? 
Also insert the signed int-variables in the union? I add this in every version

  public : union {
    uint64_t data64        ; // Caution: subject to endianness
    int64_t data_s64        ; // Caution: subject to endianness <<
    uint32_t data32 [2]    ; // Caution: subject to endianness
    int32_t data_s32 [2]    ; // Caution: subject to endianness <<
    uint16_t data16 [4]    ; // Caution: subject to endianness
    int16_t  data_s16 [4]  ; // Caution: subject to endianness <<
...

pierremolinaro, 2021-10-01:
Hello,

That is good idea, I will make this addition in my 7 CAN librairies.

tomtom0707, 2021-12-05:
Hello Pierre,

I use your library for a MCP2518 with ESP32. It works very well. Thanks.

Your description says:

Resetting the MCP2517FD can only be done by software, by sending a RESET command through the SPI.

I would now like to bring the MCP2518 to a defined state during runtime, i.e. send a reset command. 

How do you do that, do you have an example?

Thanks.

Thomas

pierremolinaro, 2021-12-07:
Hello Thomas,

You need an "end" function that resets the MCP2517FD. Unfortunatly, I haven't implemented yet.

Technically, the function just issue the RESET command, deallocate the transmitting and receiving buffer, and disable interrupt pin.

I can write this function in a few days.

Best Regards,

Pierre

tomtom0707, 2021-12-07:
Hello Pierre,

that would be great if you could add this reset/end- function in the next time :)

Thanks.

 

Best regards

Thomas

pierremolinaro, 2021-12-11:
Hello Thomas,

I have published the release 2.1.9 that contains the end method.

Best Regards,

Pierre

closed 2021-12-11 by pierremolinaro

tomtom0707, 2021-12-17:
Hello Pierre.

Thanks. 

 

Another question...

I use the MCP2518FD in normal CAN 2.0 mode with an ESP32. I use your library acan2517 for this. I don't use an interrupt, because basically only sending is interesting for me.

If I use the MCP with a crystal 20MHz, everything is fine, no mistake. 

I was too stingy for the crystal and use a pin from the ESP32 for a 4MHz clock. In addition, the setting of the PLL x10, so that I would also have all speeds available for a possible CAN FD use.

That works too and I was happy ... I thought.

Because errors occur with a precise CAN-Bus scan, see the picture. Data is being sent that has never been sent, marked in the red frame.

Isolated bit errors in the ID (always in the same place) lead to incorrectly sent ID, see colored arrows. In addition, Remote_Frame are also sent, probably the same problem in a different bit position, because they were never sent.

My guess was as follows, the ESP32's clock is not (always) clean. Now I use a separate clock crystal oscillator (7C-4.000MBA-T), but the errors remain the same.

Do you have an opinion on this matter?

A second question, do you also use interrupts for sending, so there are advantages to activating the interrupt pin for sending? Maybe the SPI-DMA mode of the ESP32 could be used sensibly?

 

Best regards

Thomas

tomtom0707, 2021-12-17:
Hello Pierre,

I've already found the solution:

OSC_4MHz10xPLL_DIVIDED_BY_2

Probably the SPI is not fast enough because I use the SPI port of the ESP32, which is still looped through the IO matrix. So it's not the quartz or oscillator that is the problem, I need the 20 Mhz system clock.

pierremolinaro, 2021-12-21:
Hello Thomas,

Glad you solved that clock problem. 

> A second question, do you also use interrupts for sending, so there are advantages to activating the interrupt pin for sending? Maybe the SPI-DMA mode of the ESP32 could be used sensibly?
Yes, I use interrupts for sending, this is required for transferring message from drive transmit buffer to MCP2518FD transmit buffer. If with your applicatio the MCP2518FD transmit buffer is never full, the sending. i nterrupt is never used.
I don't know how SPI transfer is implemented on ESP32, with or without DMA. The library is written for any Arduino compatible SPI. Sure that it could be tuned for a particular micro-controller such as ESP32, but my goal is to support the largest number of microcontrollers.

Note that giihub never transmits the attached files of a mail, so unfortunatly I can't see yout attached picture.

Best Regards,

Pierre

tomtom0707, 2021-12-21:
Hello Pierre,

thank you :) 

The attached picture only displayed the wrong messages during the CAN-BUS scan. Errors result by senting (one in about 10,000 correct messages) that are only noticeable when the bus traffic is completely recorded.

 

So here again specifically. First, the INT pin is used when new messages arrive, i.e. for reception. Second, for sending, but only when the buffer in the MCP2518 is full. What happens when the buffer is full if no INT pin is used? 

Ok, you can use in the ESP32  SPI-interrupt as well as SPI-DMA for sending. But I understand that it should be a driver for every controller with its own peculiarities. But not only that, there is the programming environment which is probably causing the problem for me. Even though the MCP is connected to the "correct" SPI pins, Arduino forces programming through the IO matrix. At least that would explain the matter, because the "correct" SPI manages 20MHz in the ESP32.

It might make sense to let the MCP run at 40MHz in order to have all CAN-FD-functions available, but to set the SPI independently, here e.g. to 15MHz if necessary. But I am initially satisfied with the solution. 

Thank you again for your work :)

 

Best Regards,

Thomas