pierremolinaro/acan2517FD #33: Trying to run on Raspberry Pico

startOfStars, 2022-11-12:
Hi

First, thank you Pierre for this great, very well docuented, library.

It runs very fine on Arduino UNO + Mikroe Click board MCP2518FD.

I'm now trying to run the library on Raspberry PICO using Arduino IDE.
I have a run severall other codes, initially intented to Arduino boards,  on Pico with success.

But with acan2517FD, no way. When I upload LoopBackDemoArduinoUnoNoInt.ino (with pins adapt) => no compiler issue, code uploaded with success but I always get 00X1 errorCode...

I have tried many many many things : with ou without interrupts, Teensy scketch (32bit), tried to adapt bit rates, etc, etc but no success...

Could someone give me a way to adapt LoopBackDemoArduinoUnoNoInt program to run on Pico.

Thanks !

pierremolinaro, 2022-11-13:
Hi,

Thanks for your encouragement!

The library should work with a Pi Pico.

If you use a Mikroe Click board MCP2518FD, are you sure that the MCP2518FD is powered with 3.3V ? For Arduino Uno, MCP2518FD should be powered with 5V, but you can damage the Pi Pico and / or MCP2518FD if the the MCP2518FD is powered with 5V and connected to a Pi Pico.

Do you assign the selected ports ? For example, if you use SPI1 :
  //--- There are no default SPI1 pins so they must be explicitly assigned
  SPI1.setSCK (MCP2518_SCK);
  SPI1.setTX (MCP2518_MOSI);
  SPI1.setRX (MCP2518_MISO);
  SPI1.setCS (MCP2518_CS);
  //--- Begin SPI1
  SPI1.begin () ;

Where
static const byte MCP2518_SCK  = 14 ; // SCK input of MCP2518 (adapt to your design)
static const byte MCP2518_MOSI = 15 ; // SDI input of MCP2518 (adapt to your design)
static const byte MCP2518_MISO = 12 ; // SDO output of MCP2518 (adapt to your design)
static const byte MCP2518_CS   = 13 ;  // CS input of MCP2518 (adapt to your design)

Regards,

Pierre

startOfStars, 2022-11-13:
Hi Pierre,

Thank you for your answer.

Yes I took attention to 3.3V/5V.

Configuration below should work, but I still get 0x1...

Thanks !

`//——————————————————————————————————————————————————————————————————————————————
//  ACAN2517FD Demo in loopback mode, for Arduino Uno, without INT pin
//——————————————————————————————————————————————————————————————————————————————

#include <ACAN2517FD.h>
#include <SPI.h>

static const byte MCP2518_SCK  = 18 ; // SCK input of MCP2518 (adapt to your design)
static const byte MCP2518_MOSI = 19 ; // SDI input of MCP2518 (adapt to your design)
static const byte MCP2518_MISO = 16 ; // SDO output of MCP2518 (adapt to your design)
static const byte MCP2518_CS   = 17 ;  // CS input of MCP2518 (adapt to your design)

//——————————————————————————————————————————————————————————————————————————————
// Very very important: put a 10kΩ resistor between CS and VDD of MCP2517FD

//static const byte MCP2517_CS  = 10 ; // CS input of MCP2517

//——————————————————————————————————————————————————————————————————————————————
//  ACAN2517FD Driver object
//——————————————————————————————————————————————————————————————————————————————

ACAN2517FD can (MCP2518_CS, SPI, 255) ; // Last argument is 255 -> no interrupt pin

//——————————————————————————————————————————————————————————————————————————————
//   SETUP
//——————————————————————————————————————————————————————————————————————————————

void setup () {
//--- Start serial
  Serial.begin (115200) ;
//--- Wait for serial (blink led at 10 Hz during waiting)
  while (!Serial) {
    delay (50) ;
  }
//----------------------------------- Begin SPI
  SPI.setSCK (MCP2518_SCK);
  SPI.setTX (MCP2518_MOSI);
  SPI.setRX (MCP2518_MISO);
  SPI.setCS (MCP2518_CS);
  SPI.begin () ;
//--- Configure ACAN2517FD
  Serial.print ("sizeof (ACAN2517FDSettings): ") ;
  Serial.print (sizeof (ACAN2517FDSettings)) ;
  Serial.println (" bytes") ;
  Serial.println ("Configure ACAN2517FD") ;
//--- For version >= 2.1.0
  ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_40MHz, 125UL * 1000UL, DataBitRateFactor::x1) ;
//--- For version < 2.1.0
//  ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_4MHz10xPLL, 125UL * 1000UL, ACAN2517FDSettings::DATA_BITRATE_x1) ;
  settings.mRequestedMode = ACAN2517FDSettings::InternalLoopBack ; // Select loopback mode
//--- Default values are too high for an Arduino Uno that contains 2048 bytes of RAM: reduce them
  settings.mDriverTransmitFIFOSize = 1 ;
  settings.mDriverReceiveFIFOSize = 1 ;
//--- RAM Usage
  Serial.print ("MCP2517FD RAM Usage: ") ;
  Serial.print (settings.ramUsage ()) ;
  Serial.println (" bytes") ;
//--- Begin
  const uint32_t errorCode = can.begin (settings, NULL) ; // Second argument is NULL -> no interrupt service routine
  if (errorCode == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Arbitration Phase segment 1: ") ;
    Serial.println (settings.mArbitrationPhaseSegment1) ;
    Serial.print ("Arbitration Phase segment 2: ") ;
    Serial.println (settings.mArbitrationPhaseSegment2) ;
    Serial.print ("Arbitration SJW:") ;
    Serial.println (settings.mArbitrationSJW) ;
    Serial.print ("Actual Arbitration Bit Rate: ") ;
    Serial.print (settings.actualArbitrationBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact Arbitration Bit Rate ? ") ;
    Serial.println (settings.exactArbitrationBitRate () ? "yes" : "no") ;
    Serial.print ("Arbitration Sample point: ") ;
    Serial.print (settings.arbitrationSamplePointFromBitStart ()) ;
    Serial.println ("%") ;
  }else{
    Serial.print ("Configuration error 0x") ;
    Serial.println (errorCode, HEX) ;
  }
}

//——————————————————————————————————————————————————————————————————————————————
//   LOOP
//——————————————————————————————————————————————————————————————————————————————

static uint32_t gSendDate = 0 ;
static uint32_t gReceiveDate = 0 ;
static uint32_t gReceivedFrameCount = 0 ;
static uint32_t gSentFrameCount = 0 ;

//——————————————————————————————————————————————————————————————————————————————

void loop () {
  can.poll () ; // No interrupt: call can.poll as often as possible
  CANFDMessage frame ;
  if (gSendDate < millis ()) {
    gSendDate += 1000 ;
    const bool ok = can.tryToSend (frame) ;
    if (ok) {
      gSentFrameCount += 1 ;
      Serial.print ("Sent: ") ;
      Serial.print (gSentFrameCount) ;
    }else{
      Serial.print ("Send failure") ;
    }
    Serial.print (", receive overflows: ") ;
    Serial.println (can.hardwareReceiveBufferOverflowCount ()) ;
  }
  if (gReceiveDate < millis ()) {
    gReceiveDate += 4567 ;
    while (can.available ()) {
      can.receive (frame) ;
      gReceivedFrameCount ++ ;
      Serial.print ("Received: ") ;
      Serial.println (gReceivedFrameCount) ;
    }
  }
}

//——————————————————————————————————————————————————————————————————————————————`

pierremolinaro, 2022-11-13:
Hi,

Yes, the configuration should work… You use a Raspberry Pi Pico, or a Raspberrey Pi Pico W ? In Pi Pico W, SPI (aka SPI0) is also used for wireless, I think its ports are set to #23, #24, … so I am not able to redirect them. Cannot you try with SPI1 ?

Pierre

startOfStars, 2022-11-13:
Below my new try with SPI1. Still no success => 0x1

(Then, just to be sure my Click board is OK, I tried again with Arduino UNO. It works.)

Thanks !

`
//——————————————————————————————————————————————————————————————————————————————
//  ACAN2517FD Demo in loopback mode, for Arduino Uno, without INT pin
//——————————————————————————————————————————————————————————————————————————————

#include <ACAN2517FD.h>
#include <SPI.h>

static const byte MCP2518_SCK  = 14 ; // SCK input of MCP2518 (adapt to your design)
static const byte MCP2518_MOSI = 15 ; // SDI input of MCP2518 (adapt to your design)
static const byte MCP2518_MISO = 12 ; // SDO output of MCP2518 (adapt to your design)
static const byte MCP2518_CS   = 13 ;  // CS input of MCP2518 (adapt to your design)

//——————————————————————————————————————————————————————————————————————————————
// Very very important: put a 10kΩ resistor between CS and VDD of MCP2517FD

//static const byte MCP2517_CS  = 10 ; // CS input of MCP2517

//——————————————————————————————————————————————————————————————————————————————
//  ACAN2517FD Driver object
//——————————————————————————————————————————————————————————————————————————————

ACAN2517FD can (MCP2518_CS, SPI1, 255) ; // Last argument is 255 -> no interrupt pin

//——————————————————————————————————————————————————————————————————————————————
//   SETUP
//——————————————————————————————————————————————————————————————————————————————

void setup () {
//--- Start serial
  Serial.begin (115200) ;
//--- Wait for serial (blink led at 10 Hz during waiting)
  while (!Serial) {
    delay (50) ;
  }
//----------------------------------- Begin SPI
  SPI1.setSCK (MCP2518_SCK);
  SPI1.setTX (MCP2518_MOSI);
  SPI1.setRX (MCP2518_MISO);
  SPI1.setCS (MCP2518_CS);
  SPI1.begin () ;
//--- Configure ACAN2517FD
  Serial.print ("sizeof (ACAN2517FDSettings): ") ;
  Serial.print (sizeof (ACAN2517FDSettings)) ;
  Serial.println (" bytes") ;
  Serial.println ("Configure ACAN2517FD") ;
//--- For version >= 2.1.0
  ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_40MHz, 125UL * 1000UL, DataBitRateFactor::x1) ;
//--- For version < 2.1.0
//  ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_4MHz10xPLL, 125UL * 1000UL, ACAN2517FDSettings::DATA_BITRATE_x1) ;
  settings.mRequestedMode = ACAN2517FDSettings::InternalLoopBack ; // Select loopback mode
//--- Default values are too high for an Arduino Uno that contains 2048 bytes of RAM: reduce them
  settings.mDriverTransmitFIFOSize = 1 ;
  settings.mDriverReceiveFIFOSize = 1 ;
//--- RAM Usage
  Serial.print ("MCP2517FD RAM Usage: ") ;
  Serial.print (settings.ramUsage ()) ;
  Serial.println (" bytes") ;
//--- Begin
  const uint32_t errorCode = can.begin (settings, NULL) ; // Second argument is NULL -> no interrupt service routine
  if (errorCode == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Arbitration Phase segment 1: ") ;
    Serial.println (settings.mArbitrationPhaseSegment1) ;
    Serial.print ("Arbitration Phase segment 2: ") ;
    Serial.println (settings.mArbitrationPhaseSegment2) ;
    Serial.print ("Arbitration SJW:") ;
    Serial.println (settings.mArbitrationSJW) ;
    Serial.print ("Actual Arbitration Bit Rate: ") ;
    Serial.print (settings.actualArbitrationBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact Arbitration Bit Rate ? ") ;
    Serial.println (settings.exactArbitrationBitRate () ? "yes" : "no") ;
    Serial.print ("Arbitration Sample point: ") ;
    Serial.print (settings.arbitrationSamplePointFromBitStart ()) ;
    Serial.println ("%") ;
  }else{
    Serial.print ("Configuration error 0x") ;
    Serial.println (errorCode, HEX) ;
  }
}

//——————————————————————————————————————————————————————————————————————————————
//   LOOP
//——————————————————————————————————————————————————————————————————————————————

static uint32_t gSendDate = 0 ;
static uint32_t gReceiveDate = 0 ;
static uint32_t gReceivedFrameCount = 0 ;
static uint32_t gSentFrameCount = 0 ;

//——————————————————————————————————————————————————————————————————————————————

void loop () {
  can.poll () ; // No interrupt: call can.poll as often as possible
  CANFDMessage frame ;
  if (gSendDate < millis ()) {
    gSendDate += 1000 ;
    const bool ok = can.tryToSend (frame) ;
    if (ok) {
      gSentFrameCount += 1 ;
      Serial.print ("Sent: ") ;
      Serial.print (gSentFrameCount) ;
    }else{
      Serial.print ("Send failure") ;
    }
    Serial.print (", receive overflows: ") ;
    Serial.println (can.hardwareReceiveBufferOverflowCount ()) ;
  }
  if (gReceiveDate < millis ()) {
    gReceiveDate += 4567 ;
    while (can.available ()) {
      can.receive (frame) ;
      gReceivedFrameCount ++ ;
      Serial.print ("Received: ") ;
      Serial.println (gReceivedFrameCount) ;
    }
  }
}

//——————————————————————————————————————————————————————————————————————————————
`

startOfStars, 2022-11-13:
I use PICO H, not W.

kholia, 2024-11-29:
Hi,

I am running into the same problem (`Configuration error 0x1`) on a Raspberry Pi Pico.

I have the `LoopBackDemoArduinoUnoNoInt` example perfectly well on the combination of Arduino Uno and Longcan CAN-FD Shieldhardware. No problems there!

Here is the complete `LoopBackDemoArduinoUnoNoInt-Pico` code listing.

```C
//——————————————————————————————————————————————————————————————————————————————
//  ACAN2517FD Demo in loopback mode, for Arduino Uno, without INT pin
//——————————————————————————————————————————————————————————————————————————————

#include <ACAN2517FD.h>
#include <SPI.h>

//——————————————————————————————————————————————————————————————————————————————
// Very very important: put a 10kΩ resistor between CS and VDD of MCP2517FD (YES!)
// Connect 3.3V to Longan CANFD module's 3.3V pin.

// Pico 2 -> Default SPI
// #define PIN_SPI0_MISO  (16u)
// #define PIN_SPI0_MOSI  (19u)
// #define PIN_SPI0_SCK   (18u)
// #define PIN_SPI0_SS    (17u)
static const byte MCP2517_CS = PIN_SPI0_SS;  // CS input of MCP2517

//——————————————————————————————————————————————————————————————————————————————
//  ACAN2517FD Driver object
//——————————————————————————————————————————————————————————————————————————————

// ACAN2517FD can(MCP2517_CS, SPI, 255);  // Last argument is 255 -> no interrupt pin
ACAN2517FD can(MCP2517_CS, SPI, 255);  // Last argument is 255 -> no interrupt pin

//——————————————————————————————————————————————————————————————————————————————
//   SETUP
//——————————————————————————————————————————————————————————————————————————————

void setup() {
  //--- Start serial
  Serial.begin(115200);
  //--- Wait for serial (blink led at 10 Hz during waiting)
  while (!Serial) {
    delay(50);
  }
  //----------------------------------- Begin SPI

  SPI.setMISO(PIN_SPI0_MISO);
  SPI.setCS(PIN_SPI0_SS);
  SPI.setSCK(PIN_SPI0_SCK);
  SPI.setMOSI(PIN_SPI0_MOSI);
  SPI.begin(false);

  //--- Configure ACAN2517FD
  Serial.print("sizeof (ACAN2517FDSettings): ");
  Serial.print(sizeof(ACAN2517FDSettings));
  Serial.println(" bytes");
  Serial.println("Configure ACAN2517FD");
  //--- For version >= 2.1.0
  ACAN2517FDSettings settings(ACAN2517FDSettings::OSC_20MHz, 125UL * 1000UL, DataBitRateFactor::x1);
  //--- For version < 2.1.0
  //  ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_4MHz10xPLL, 125UL * 1000UL, ACAN2517FDSettings::DATA_BITRATE_x1) ;
  settings.mRequestedMode = ACAN2517FDSettings::InternalLoopBack;  // Select loopback mode
  //--- Default values are too high for an Arduino Uno that contains 2048 bytes of RAM: reduce them
  settings.mDriverTransmitFIFOSize = 1;
  settings.mDriverReceiveFIFOSize = 1;
  //--- RAM Usage
  Serial.print("MCP2517FD RAM Usage: ");
  Serial.print(settings.ramUsage());
  Serial.println(" bytes");
  //--- Begin
  const uint32_t errorCode = can.begin(settings, NULL);  // Second argument is NULL -> no interrupt service routine
  if (errorCode == 0) {
    Serial.print("Bit Rate prescaler: ");
    Serial.println(settings.mBitRatePrescaler);
    Serial.print("Arbitration Phase segment 1: ");
    Serial.println(settings.mArbitrationPhaseSegment1);
    Serial.print("Arbitration Phase segment 2: ");
    Serial.println(settings.mArbitrationPhaseSegment2);
    Serial.print("Arbitration SJW:");
    Serial.println(settings.mArbitrationSJW);
    Serial.print("Actual Arbitration Bit Rate: ");
    Serial.print(settings.actualArbitrationBitRate());
    Serial.println(" bit/s");
    Serial.print("Exact Arbitration Bit Rate ? ");
    Serial.println(settings.exactArbitrationBitRate() ? "yes" : "no");
    Serial.print("Arbitration Sample point: ");
    Serial.print(settings.arbitrationSamplePointFromBitStart());
    Serial.println("%");
  } else {
    Serial.print("Configuration error 0x");
    Serial.println(errorCode, HEX);
  }
}

//——————————————————————————————————————————————————————————————————————————————
//   LOOP
//——————————————————————————————————————————————————————————————————————————————

static uint32_t gSendDate = 0;
static uint32_t gReceiveDate = 0;
static uint32_t gReceivedFrameCount = 0;
static uint32_t gSentFrameCount = 0;

//——————————————————————————————————————————————————————————————————————————————

void loop() {
  can.poll();  // No interrupt: call can.poll as often as possible
  CANFDMessage frame;
  if (gSendDate < millis()) {
    gSendDate += 1000;
    const bool ok = can.tryToSend(frame);
    if (ok) {
      gSentFrameCount += 1;
      Serial.print("Sent: ");
      Serial.print(gSentFrameCount);
    } else {
      Serial.print("Send failure");
    }
    Serial.print(", receive overflows: ");
    Serial.println(can.hardwareReceiveBufferOverflowCount());
  }
  if (gReceiveDate < millis()) {
    gReceiveDate += 4567;
    while (can.available()) {
      can.receive(frame);
      gReceivedFrameCount++;
      Serial.print("Received: ");
      Serial.println(gReceivedFrameCount);
    }
  }
}

//——————————————————————————————————————————————————————————————————————————————
```

There must be something going wrong with the SPI communication it seems.

kholia, 2024-11-29:
> There must be something going wrong with the SPI communication it seems.

All good now - the library works great on the Pico! The problem was that I was not powering the Longan's shield correctly. The `5V` point of the shield must be connected to `3.3V` point of the Pico!

Thank you for this awesome library! <3