pierremolinaro/acan2517FD #37: Problem Recieving messages with MCP2517FD Click

GLADIS-Kernel, 2023-04-06:
Hi pierre,

First of all, thank you for your development of such a powerful library regarding CANFD. 
My issue here is that I can't recieve CAN2.0 messages using a innomaker usb2can device. The code and hardware work perfectly when using external or internal loopback mode but with NormalFD mode i can't seem to recieve messages throurgh the db9 connector. 

I'm using a mikrobus arduino uno click shield connected to an arducam ESP32. I've tried changing the VIO SEL jumper to 5v but hasn't helped. I've also added a 120ohm resistor to the innomakers output to have a 120 resistor on both ends. Didn't help either.

I'm attaching both the recieve error passive and the code. Thanks !!!

[image]

#include <Ticker.h>

#include <FS.h>
#include <SPI.h>
#include <ACAN2517FD.h>
#include <SD.h>

//——————————————————————————————————————————————————————————————————————————————
//  ACAN2517FD (ESP32) + SD Datalogger + Emulator
//——————————————————————————————————————————————————————————————————————————————

#ifndef ARDUINO_ARCH_ESP32
  #error "Select an ESP32 board"
#endif

// Define EMULATOR for simulations
//#define EMULATOR

//——————————————————————————————————————————————————————————————————————————————
//  Libraries
//——————————————————————————————————————————————————————————————————————————————

static const byte MCP2517_SCK  = 25 ; // SCK input of MCP2517
static const byte MCP2517_MOSI = 27 ; // SDI input of MCP2517
static const byte MCP2517_MISO = 26 ; // SDO output of MCP2517
static const byte MCP2517_CS  = 0 ; // CS input of MCP2517
static const byte MCP2517_INT = 13 ; // Interruption PIN of MCP2517

//——————————————————————————————————————————————————————————————————————————————
//  Global Variables: ACAN2517FD Driver object + SPI Class + unint8_t's
//——————————————————————————————————————————————————————————————————————————————

SPIClass hspi(HSPI);

ACAN2517FD can (MCP2517_CS, hspi, MCP2517_INT) ; 

CANFDMessage frame ;

CANFDMessage FIFO[256];

uint8_t in, out = 0;

Ticker blinker;

const float blinkerPace = 0.005;  //seconds

static uint32_t gWriteFrameCount = 0 ;
static uint32_t gSentFrameCount = 0 ;

File dataFile;
File readFile;

//——————————————————————————————————————————————————————————————————————————————
//   SETUP
//——————————————————————————————————————————————————————————————————————————————

void setup () {

  delay(2000);

//--- Start serial
  Serial.begin (115200) ;

  
//--- Begin SPI
 
  hspi.begin (MCP2517_SCK,MCP2517_MISO,MCP2517_MOSI,MCP2517_CS) ;

//--- Configure ACAN2517FD
  Serial.print ("sizeof (ACAN2517FDSettings): ") ;
  Serial.print (sizeof (ACAN2517FDSettings)) ;
  Serial.println (" bytes") ;
  Serial.println ("Configure ACAN2517FD") ;
  
//--- For version >= 2.1.0
  ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_40MHz, 500UL * 1000UL, DataBitRateFactor::x1);

  #ifdef EMULATOR
        settings.mRequestedMode = ACAN2517FDSettings::ExternalLoopBack;
  #else
        settings.mRequestedMode = ACAN2517FDSettings::NormalFD;
  #endif
  
  
//--- RAM Usage
  Serial.print ("MCP2517FD RAM Usage: ");
  Serial.print (settings.ramUsage ());
  Serial.println (" bytes");

//----------------------------------- Append filters
  ACAN2517FDFilters filters;
  //filters.appendFrameFilter (kStandard,0x120,receiveFromFilter1) ; // Filter ID = 0x120, Water and Intake air temp
  //filters.appendFrameFilter (kStandard,0x100,receiveFromFilter2) ; // Filter ID = 0x100, RPM
  //filters.appendFilter (kStandard,0x001,0x001,receiveFromFilter2); // Filter Impar
  //filters.appendFilter (kStandard,0x001,0x000,receiveFromFilter3); // Filter Par
  filters.appendPassAllFilter (receiveFromFilter0);  // Filter ALL
  
 
//----------------------------------- Filters ok ?
  if (filters.filterStatus () != ACAN2517FDFilters::kFiltersOk) {
    Serial.print ("Error filter ") ;
    Serial.print (filters.filterErrorIndex ()) ;
    Serial.print (": ") ;
    Serial.println (filters.filterStatus ()) ;
  }
/*
  settings.mISOCRCEnabled=false;
  settings.mDriverTransmitFIFOSize = 1 ;
  settings.mDriverReceiveFIFOSize = 1 ;

*/
//--- Begin CAN
  const uint32_t errorCode = can.begin (settings, canISR,filters) ;

  //const uint32_t errorCode = can.begin (settings, canISR) ;
  //Serial.println(errorCode);

  if (errorCode == 0) {
    Serial.println(settings.CANBitSettingConsistency(),HEX);
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Arbitration Phase segment 1: ") ;
    Serial.println (settings.mArbitrationPhaseSegment1) ;
    Serial.print ("Arbitration Phase segment 2: ") ;
    Serial.println (settings.mArbitrationPhaseSegment2) ;
    Serial.print ("Arbitration SJW:") ;
    Serial.println (settings.mArbitrationSJW) ;
    Serial.print ("Actual Arbitration Bit Rate: ") ;
    Serial.print (settings.actualArbitrationBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact Arbitration Bit Rate ? ") ;
    Serial.println (settings.exactArbitrationBitRate () ? "yes" : "no") ;
    Serial.print ("Arbitration Sample point: ") ;
    Serial.print (settings.arbitrationSamplePointFromBitStart ()) ;
    Serial.println ("%") ;

    
    //--- Start Ticker - CAN Sender
    #ifdef EMULATOR
    blinker.attach(blinkerPace,blink);
    #endif

  
    //--- Start SD
    if(!SD.begin()){
        Serial.println("Card Mount Failed");
        return;
    }

  //createDir(SD,"/CAN_MOTO");
  //writeFile(SD, "/CAN_MOTO/Data.dat", "This is the file which is gonna save the CAN Messages\n");
  dataFile = SD.open("/datalog.dat", FILE_WRITE);
  if(!dataFile){
    Serial.println("Error opening file");
  }
  }
  else{
    Serial.print ("Configuration error 0x") ;
    Serial.println (errorCode, HEX) ;

    //Restart if CAN doesnt begin correctly
    ESP.restart();
  }

  
  //fillDATA();
  //if(can.tryToSend(frame)){Serial.println("SENT");}
}

//——————————————————————————————————————————————————————————————————————————————
//   Interruption Function
//—————————————————————————————————————————————————————————————————————————————-

void canISR () {
  
  can.isr () ;

}

//——————————————————————————————————————————————————————————————————————————————
//   BLINK() CAN SENDER
//——————————————————————————————————————————————————————————————————————————————

void blink(){

    // Send CAN Message

    fillDATA();
    delay(400);
   
    if (can.tryToSend(frame)) {
      gSentFrameCount ++ ;
      Serial.print ("Sent: ") ;
      Serial.println (gSentFrameCount) ;
      
    }else{
      Serial.println ("Send failure") ;
      //if (frame.isValid()) {
        //Serial.println("Is Valid");
      //}
      Serial.print(frame.id);
      Serial.println(frame.len);
      //Serial.println(frame.data[3]);
      //appendFile(SD,"/CANMessages4/Data.txt","send failure");
      
    }
}

//——————————————————————————————————————————————————————————————————————————————
//   LOOP
//——————————————————————————————————————————————————————————————————————————————

char msg[276], msgdata[276];

struct datastore {
    unsigned long timestamp;
    uint16_t id;
    unsigned char msg_data[8];
};

void loop () {

    // When Received (in != out) --> Pop out of FIFO
    while(can.dispatchReceivedMessage()){
    if (in != out){
      
       struct datastore myData;
       struct datastore readData;
       
       //Initialize all the data from FIFO[out] = Frame recieved from interruption
       
       
       myData.id=FIFO[out].id;
        
       unsigned int len=0;
       len = FIFO[out].len;

       myData.timestamp = millis();

       //Serial.println(FIFO[out].type);

       //uint8_t data[64]; // Fixed to maximum length of CANFD-Data  (64 bytes)             
       for (int i = 0; i<len;i++)  // Array index --> From position 0 to (lenght-1)
        {
          myData.msg_data[i]=FIFO[out].data[i];
        }

        dataFile.write((const uint8_t *)&myData, sizeof(myData));

        // Print array to char 
        PrintHex8(myData.msg_data,len,msgdata);
       
         //Concatenates chars
        sprintf (msg,"%d %X %d ",myData.timestamp,myData.id,sizeof(myData.msg_data));
        strcat(msg,msgdata); 
        Serial.println(msg);

        out++;

    #ifdef EMULATOR
    if (gWriteFrameCount == 100){
      blinker.detach();
      dataFile.close();
      ReadMsg();
      //Stop();
      }
    #endif
    
     gWriteFrameCount ++ ;
     
    }
    }
    
  
}

//——————————————————————————————————————————————————————————————————————————————

pierremolinaro, 2023-04-07:
Hello,

Do not set the VIO jumper to +5V : the ESP32 is not +5V compliant, you can damage it with 5V levels.

So you send frames with arduino uno click shield using ACAN2517FD, and you want to receive them on a innomaker usb2can device.

I have several questions.

1) You are sending frames with this code.

> CANFDMessage frame ;
and in the loop :
> if (can.tryToSend(frame)) {

So the frames are sent using default values. But default format is CANFD with bit switch rate, and is not a valid 2.0B frame, so the innomaker usb2can device detect an error.

Try :
frame.type = CAN_DATA ; // CAN 2.0B data frame
if (can.tryToSend(frame)) {
2) How the CAN network is wired. Make sure you have CANH, CANL, and a common ground.

Regards,

Pierre

GLADIS-Kernel, 2023-04-08:
Hi Pierre,

Thanks for the answer ! I've changed the VIO jumper and implemented your code using CAN_DATA message types but still, i can't send or recieve messages that are readable, error passive still raises.

    // Send CAN Message
    CANFDMessage frameSEND ;
    frameSEND.type=CANFDMessage::CAN_DATA;
    fillDATA(frameSEND);
    delay(500);
   
    if (can.tryToSend(frameSEND)) {
      gSentFrameCount ++ ;
      Serial.print ("Sent: ") ;
      Serial.println (gSentFrameCount) ;
      
    }

fillDATA fills the frame with id and random 8 bytes of data.

Maybe I'm having problems with the wiring, but right now testing with the multimeter, i'm having proper connection with CANH, CANL, and GND in both ends of the nodes. I'll try my best checking again. 

Another cause could be the ATA6563 pyshical layer, but it seems to work fine altough its always in recesive state.

thanks !!

kukumagi, 2024-03-23:
Did you get it to work?
I have MCP2518FD board and it wont work unless I disconnect 5v supply from MCP and then reconnect, then it runs fine, but how could I get the same result with code?

GLADIS-Kernel, 2024-03-26:
Hi, I ended up changing the whole HW with a new WIO ESP32 CAN Dev Kit from seeed studio.
After many tryouts i ended up always with the can bus in a rececive state so i decided to start from scratch. These new HW doesn't implement Pierre's library but i don't think the problem was the code. I guess I was missing something from the wiring or HW HAL drivers.

kukumagi, 2024-03-26:
> Hi, I ended up changing the whole HW with a new WIO ESP32 CAN Dev Kit from seeed studio. After many tryouts i ended up always with the can bus in a rececive state so i decided to start from scratch. [...]

Thanks for the response.
I guess I will try some different code options and if that doesnt work then I could add a relay to 5v line to reset the 6560 chip :D