pierremolinaro/acan2517FD #39: Problem with arduino due

Maaz-Sansare, 2023-07-08:
I encountered a timeout error while attempting to run one of your example codes, specifically the loopback code, on an Arduino Due. In order to investigate the issue further, I decided to write a small program to configure and trigger the GPIO of the click board, and that seemed to work without any problems.

Now, I'm wondering if there are any specific configurations that I should be aware of when trying to configure CANFD using the Arduino Due. Could you provide me with some guidance on this matter? Alternatively, is it possible that the issue I'm experiencing is a result of a bug in the code or the system?

pierremolinaro, 2023-07-08:
Hello,

I cannot build the same configuration, as I have no Arduino Due.

I have just understood that there is a small bug in my library. I have taken a look the MCP251863 datasheet DS20006624A). Section 4 is SPI clock speed, it should be less than or equal to 0.85 * (FSYSCLK/2). But my library is based on MCP2518FD datasheet ( DS20000000A) where section 4 says "The frequency of SCK has to be less than or equal to half the frequency of SYSCLK".

Can you change a line of the ACAN2517FD.cpp file in the ACAN2517FD library :
line 351 :
  mSPISettings = SPISettings (inSettings.sysClock () / 2, MSBFIRST, SPI_MODE0) ;
change to :
  mSPISettings = SPISettings ((inSettings.sysClock () * 2) / 5, MSBFIRST, SPI_MODE0) ;

Can you try this ?

Pierre

Maaz-Sansare, 2023-07-09:
same issue

Maaz-Sansare, 2023-07-09:
May this help. I tried the ACAN2517 library loopback example code, and after a few resets, it started working with the same SPI settings as ACAN2517FD.
mSPISettings = SPISettings (inSettings.sysClock () / 2, MSBFIRST, SPI_MODE0) ;