pierremolinaro/acan2517FD #40: Additinal features

pilotak, 2023-08-01:
First of all what a perfect library, thanks! i love the docu :heart:

It would be perfect to add following features

# More operation modes

110 = Set Normal CAN 2.0 mode
001 = Set Sleep mode _(this one is tricky to call in begin I believe so it would be nice to have dedicated function for sleep)_

https://github.com/pierremolinaro/acan2517FD/blob/a17063059c9d26612879080402eb98283b608253/src/ACAN2517FDSettings.h#L43-L48

ideally there should be defined only one enum for modes, at the moment there are two - `OperationMode` and `RequestedMode`

# Ability to use GPIOs
specifically use the `XSTBYEN`  function, which will save one GPIO on MCU - together with added sleep function this is perfect duo

---

I can submit a PR and test physically after discussion

pierremolinaro, 2023-08-05:
Hello,

Thank you very much for your thanks!

I'm going to add the features you asked for, and I'll do a release in a few days.

Best Regards,

Pierre

pierremolinaro, 2023-08-09:
Hello,

I have added the features, currently the new release is not available from Arduino IDE, you can get it at:
	https://github.com/pierremolinaro/acan2517FD

I have also updated the doc.

OperationMode and RequestedMode have been merged, now it is ACAN2517FDSettings::OperationMode ;

I have added to the ACAN2517FD class the following methods: gpioSetMode, gpioWrite, gpioRead and configureGPIO0AsXSTBY. Currently, I have no hardware to check theses methods.

Can you test them ?

If they work, I will do a release.

Best Regards,

Pierre

pilotak, 2023-08-09:
Thanks a lot, sure I will test it and you know

pilotak, 2023-08-12:
putting in into sleep works, power power consumption drops, can bus is off

the problem is when you want to wake up - setting a mode `can.setOperationMode(ACAN2517FDSettings::Normal20B);` when previous was sleep `can.setOperationMode(ACAN2517FDSettings::Sleep); ` does not wake up the chip, even when the whole setup runs again (processor reset) - gives `Configuration error 0x2`. Only power cycle the can chip cancels the sleep mode.

Unfortunately `configureGPIO0AsXSTBY` doesn't work and and GPIO either but this could be because `PM0`/`PM1` is not set as GPIO and remains `INT` by default

pilotak, 2023-08-12:
I have submitted a PR for GPIO fix #42

pilotak, 2023-08-13:
perfect! thanks a lot, all works well

pierremolinaro, 2023-08-14:
Thanks, I publish the release.