pierremolinaro/acan2517FD #41: Trying to run on Arduino Giga R1

startOfStars, 2023-08-07:
Hi everybody,

I'm trying to run "LoopBackDemoArduinoUnoNoInt" example on Arduino Giga R1 (I was able to run the ACAN2517FD library on Rapsberry Pico).

Each time I open the Serial Monitor the sketch freezes just after "MCP2517FD RAM Usage: 2016 bytes" (Arduino Giga R1 red led lights up).

It seems that sketch freezes on line : "const uint32_t errorCode = can.begin (settings, NULL)"

Is there a chance that ACAN2517FD works on Arduino Giga R1 or shall I stop my tests ?

Thanks !
(and thanks again to Pierre for this library !)

pierremolinaro, 2023-08-07:
Hello,

Thank you for your thanks!

The library should run on any microcontroller that has a SPI. You call begin with no interrupt routine, How do you declare the can object ? The constructor last argument is the int pin number, it should be 255 as there is no interrupt routine.

Unfortunatly, I have no Arduino Giga R1, so I cannot try.

However, the MCP2518FD is a temperamental component, I have experencied difficulties with some boards.

Hope that helps,

Pierre

startOfStars, 2023-08-08:
Hi Pierre

Thank you for your quick answer.

I took care about what you said, no problem on that.

"The library should run on any microcontroller that has a SPI" => that's what I was thinking about that's why I don' understand whay do not find the way to run on Giga.

Since yesterday I made severall tests. I use your library since severall months that's why I'm aware about what I have to be carrefull about (int / on int / Hz quartz frequency and so one).

I do'nt understand, I double checked my connections, also tried with Mikroe Click : ok on Uno and Raspberry Pico but fail with Giga...

I'm pretty sure there is a problem with SPI connection but I don't understand why. I'm running Ethernet with UDP on Giga wieth SPI connection on others projects with success.

Best regards and thanks !

startOfStars, 2023-08-10:
After more investigations, I found that calls of noInterrupts () and interrupts () in ACAN2517FD.cpp causes Arduino Giga R1 crashs.
When commenting these parts of sketch, the sketch runs well.

Arduino Giga R1 is supposed to be compatible with these inetrrupt functions.

I don't find what I'm missing.

pierremolinaro, 2023-08-10:
Strange, which lines did you comment on?

startOfStars, 2023-08-10:
In file ACAN2517FD.cpp, all parts as :
`
    #ifdef ARDUINO_ARCH_ESP32
      taskDISABLE_INTERRUPTS () ;
    #else
      noInterrupts () ;
    #endif
`
and 
`
    #ifdef ARDUINO_ARCH_ESP32
      taskENABLE_INTERRUPTS () ;
    #else
      interrupts () ;
    #endif
`
Thanks!

startOfStars, 2023-08-12:
To be more precised I made _search/replace_ in ACAN2517FD.cpp:
`noInterrupts () ; = > delay(0); // noInterrupts () ;`

and then :

`interrupts () ; => delay(0); // interrupts () ;`

Just retried, if not modified => OS crash / if modified as above => success (CAN message well sent)

pierremolinaro, 2023-08-14:
Ok, I think that delay(0) is a no-operation. Removing noInterrupts / interrupts is safe for you, as you have configured the library for not using interrupts.

startOfStars, 2023-08-16:
Hi

I'm motivated to make this amazing library to work on Arduino Giga R1 🙂

So to resume :

- I'm working (for the moment) with the example sketch LoopBackDemoArduinoUnoNoInt
- noInterrupts / interrupts are supposed to work with Arduino Giga R1. See post#4 of Arduino Team member here : https://forum.arduino.cc/t/cli-sei-and-pind-equivalent-commands-on-giga-r1-mbed-os/1111448/3
- I have tested to run simple sketch with noInterrupts / interrupts functions on Giga R1 => no crash

After more investigations I found that this is noInterrupts () function only that causes crash. When I comment only this one (letting interrupts () ; ) it works. So this is noInterrupts () that causes crash.

How shall I continue my investigations ? How noInterrupts () could make mbed crashes in this library ?

Thanks !

pierremolinaro, 2023-08-18:
Hello,

It's an excellent idea and I encourage you to persevere.

I have two suggestions.

First, I have (re)discovered a long time pull request (#35) about a wrong attach interrupt ordrer. I have merged this pull request, can you try with the library on GitHub (https://github.com/pierremolinaro/acan2517FD) ?

Then, if it does not work, you can add several Serial.print in the begin method for tracking where things go wrong.

Pierre

startOfStars, 2023-08-28:
Hi Pierre

Unfortunatly, new try of the library with attach interrupt order updated does not solve the issue...

I made an other test :

1. First I replaced all noInterrupts() by delay(0) in ACAN2517FD.cpp filte => success
2. I added two Serial.print like this in begin method :

`  //----------------------------------- Reset MCP2517FD (always use a 800 kHz clock)
	Serial.println("before reset");
    reset2517FD () ;
	Serial.println("after reset");`

I checked, "before reset" and "after reset" are well printed.

3. I put back noInterrupts() in :
`void ACAN2517FD::reset2517FD (void) {
  mSPI.beginTransaction (mSPISettings) ; // Check RESET is performed with 800 kHz clock
    #ifdef ARDUINO_ARCH_ESP32
      taskDISABLE_INTERRUPTS () ;
    #else
      noInterrupts () ; // <=
    #endif
      assertCS () ;
        mSPI.transfer16 (0x00) ; // Reset instruction: 0x0000
      deassertCS () ;
    #ifdef ARDUINO_ARCH_ESP32
      taskENABLE_INTERRUPTS () ;
    #else
      interrupts () ;
    #endif
  mSPI.endTransaction () ;
}`

**=> Crash**

So it's sure the issue is due to noInterrupts()

5. I ran a very simple sketch on Giga :
`void setup() {}

void loop() {
  noInterrupts();
  // critical, time-sensitive code here
  interrupts();
  // other code here
}`

**=> no problem**

I don't understand.

Thanks!

emagnetto, 2024-02-14:
Hi! TEsting also on Giga R1. Same issue as mentioned here. Under analysis now. 
@startOfStars  what CAN controller are you using? MCP2518FD ? Purchased MCP251863 so as to start my tests.

Are you also trying work on FDCAN Giga native ports?

pierremolinaro, 2024-02-17:
Hello,

I cannot test on Giga R1, I do not have this board, that is quite expensive.

The MCP251863 is a MCP2518FD with an embedded transceiver, so I think you will get the same problems.

The STM32H747 microcontroller contains two CANFD controllers, I think handling them is quite similar to ACANFD_STM32, but this library is not compatible with Giga R1.

Pierre

emagnetto, 2024-02-20:
Pierre, thank's for your response.
The Giga is an interesting board, specially whenever you want to be close to a Mega layout.
I received it a couple of weeks ago. I agree, a little bit expensive ...
I think the issue has to do with the mbed layer, but had no time to work on the issue yet.

About MCP251863 and MCP2518FD. I know both. I studied a them before purchasing. In fact, I bought the former (a nice CLICK board from MIKRO, easy to connect to any Arduino), as costs almost half than the latter (with same transiever). Still waiting for them ... will arrive in a couple of days.

I'm designing a custom system for flight simulators instruments and commands, based on a "powerfull" master node (by now, a GIGA, but may be other STM32, depending on my future complete requirements), and several satellite, slave nodes (RP2040, and perhaps others), with FDCAN for connectivity. 

I started analyzing acanfd_stm32 last week. Although MCU is similar on Giga, lib is not compatible, as you said.
On the other hand, I was not able to find any FDCAN library for Giga nor Portenta ...
Hence, I'm still on moving sands ... trying to fix hw or sw decisions and start from there.
That's why I reached your libraries! They will give me a common ground to start from.

In fact, I also need a higher level protocol (like CANOpen), although it will be an overkill for my needs.
Thinking on implementing just a simplified subset.

pierremolinaro, 2024-02-24:
Hello,

I understand why you hase chosen a GIGA R1.I 've had a look at the features of this card, and it's indeed interesting. Dual-core offers additional possibilities. I've decided to order one, which I should receive next week. I'll try to port the ACANFD_STM32 library to this card, and to understand why the acan2517FD crashes.

I don't know CANOpen, so I can't help you there.

Pierre

emagnetto, 2024-02-24:
Wow! I'll wait for your outcome. Not sure how complex it can be. Although it has an STM32 mcu, whole "arduino" and mbed stuff may impose some difficulties.
Today I received some CAN controllers, so can start playing with it. Will also analyze the crash.

I've been studying CANOpen for a couple of weeks on my spare time, trying to get a  higher level layer. 
Many web sources and doc. But hard to turn to real working code.
https://github.com/CANopenNode/CANopenNode has a nice starting point, although not FD compatible.
My goal: a (sort of) CANOpen FD implementation on top of ACAN2517FD and ACANFD_STM32

emagnetto, 2024-03-01:
Just an insight on the nointerrupt() issue.

Made additional tests, like @startOfStars  on Aug, 28th , on reset2517FD().
The issue seems to be on SPI transfers (mSPI.transfer16 (0x00) ; )  whenever noInterrupts() has been called.
Hence, tested code below. 
It crashes with red blinking led. (mbed crash)

`#include <SPI.h>

const int CS_1 = 7;

void setup() {
	pinMode(CS_1, OUTPUT);
	Serial.println(" START");
	SPI.begin();

	digitalWrite(CS_1, LOW);

	noInterrupts();
	SPI.transfer(0x00);
	interrupts();
	digitalWrite(CS_1, HIGH);

	Serial.println(" END");
}
`

pierremolinaro, 2024-03-02:
Hello,

I have received my Giga R1 board, and I have reproduced the crash you describe.

let me a few days to understand what is happening.

Pierre

pierremolinaro, 2024-03-02:
I think that noInterrupts and SPI.transfer cannot be used together on Arduino GIGA R1.

The Arduino GIGA R1 firmware is MBED based, so it is not fully compatible with usual API. In MBED, it seems that  SPI.transfer requests a lock that requires the interrupt are enabled.

So ACAN2517FD should be modified to handle the case of GIGA R1. I think the way is to create an utility task, as for ESP32 under RTOS.

The good news is that I've started adapting the ACANFD_STM32 library to GIGA R1. Currently it runs in loopback mode, but I didn’t not check yet the bitrate.

Pierre

emagnetto, 2024-03-02:
Hi! I did same analysis about GIGA and MBED. 
In fact, already made some changes to support GIGA, in line with what you proposed, as an initial testbed.

Anyway, the second approach is smarter. In fact yesterday I downloaded the STM32H747 Ref Manual (more than 3000 pages!) and looked at FDCAN specs and analyzed a little bit ACANFD_STM32 code.
Now that I know where to look, it´s not so scary as initially thought ... Your code is modular and very neat. And as you mentioned before, Portenta and Giga shares a very similar MCU as one of the NUCLEO (H743ZI2) you developed for ... In fact, checking both Ref Manuals, FDCAN implementation seems to be the same.
It´s great you have already running on loop!

By the way, I was studing a little bit Saleae products (I want to purchase a Logoc Analyzer), and found you developed an FDCAN plugin for it!

pierremolinaro, 2024-03-03:
Hello!

I have published a first release of the ACANFD_GIGA_R1 library, at https://github.com/pierremolinaro/acanfd-giga-r1. It is not available yet on Arduino library manager.

The default CANCLOCK is 60 MHz, so the 8 MBit/s data rate cannot be achieved (because 60 is not a multiple of 8).

Salae logic analyzers are quite expensive, and I hesitated for a long time before buying one. I had previously bought a Chinese model, and found that its clock tended to drift, so that the durations were not exact! The Salae Logic Pro 8 I have doesn't have this defect, it works perfectly, and I find the Logic software simple and efficient.  I also wrote CAN and CAN FD analyzers because the ones I found didn't satisfy me.

For ACAN2517FD, I have to adapt the code to the GIGA R1. Currently, the library handles two configurations, one for ESP32, one for regular Arduino. I need to create a third one for the GIGA R1. I think this requires a new architecture for handing this nicely.

Pierre

emagnetto, 2024-03-03:
Pierre, thanks for this great package! I´ll test it tomorrow.
I´m ending my vacations (I took time from it to start studying CAN ... My wife was not happy), and had not enough time yet to go deeper on coding and testing.
What I will also test tomorrow is this code (acan2517FD) and my new hardware with Pi Pico (RP2040).
Also analyzed two CANOpen codes (Easy-CANOpen and CANOpenNode), so as to build this protocol (or part of it) on top of your CAN libs. Easy-CANOpen seems to be easier to adapt. I´m not sure how much time I will have now that I´m back to work ... But will let you know the results, and of course, if I have any issue or suggestions, will open an issue!