pierremolinaro/acan2517FD #43: ESP32-S3 + MCP2518FD

jblazeg, 2023-11-23:
Hi, 

I use a custom board and I get "Configuration error 0x10000" - which  means timeout initialising the SPI device.
[image]

But when I disconnect the other can bus device (works - verified) it initializes correctly, but still doesnt send any data (CAN TX line is permanently high, also no CAN receive):
[image]

Anything else I should capture or add files? CAN Select has a 10K pullup resistor. I use a 40MHz resonator with 22pF capacitors

extract from the code:
```
SPIClass SPI2(HSPI);

const uint8_t SOFT_SCK_PIN  = 19;
const uint8_t SOFT_MISO_PIN = 20;
const uint8_t SOFT_MOSI_PIN = 21;
#define SELECT 17 
ACAN2517FD can2(SELECT, SPI2, 255)

setup() {

SPI2.begin(SOFT_SCK_PIN,SOFT_MISO_PIN,SOFT_MOSI_PIN);
SPI2.setFrequency(5000000);

ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_40MHz_DIVIDED_BY_2, 125 * 1000, DataBitRateFactor::x4) ;
settings.mRequestedMode = ACAN2517FDSettings::NormalFD; 

const uint32_t errorCode = can2.begin (settings, NULL) ;
}
```

Cheers

pierremolinaro, 2023-11-27:
Hello,

I'm afraid I can't help you, the MCP2518 is a temperamental component, and many users have reported difficulties with initialization. Error 0x10000 means that the MCP2518 does not go into configuration mode.

You get different behaviour depending on whether the MCP2518 is connected to the bus or not. Is the electrical ground the same?

Pierre

jblazeg, 2023-11-27:
Yes, the behaviour is different. With can bus traffic, the chip does not initialise. But it does not receive/send anything even if it initialises correctly.

Common ground is generally not needed for can bus (differential pair) but I tried with common ground and without - the problem is the same.

VergilGao, 2023-12-26:
hello, di

kukumagi, 2024-02-27:
> ![tb_image_share_1703575827964.jpg.png](https://private-user-images.githubusercontent.com/8655163/292826674-9983767b-1e52-4fdd-bbd6-1708bc15df45.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNjE3NzYsIm5iZiI6MTcwOTA2MTQ3NiwicGF0aCI6Ii84NjU1MTYzLzI5MjgyNjY3NC05OTgzNzY3Yi0xZTUyLTRmZGQtYmJkNi0xNzA4YmMxNWRmNDUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjdUMTkxNzU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2M0N2QwOTVlMmY2NTA5M2FlYmQzOWY5NmE5OTFkYjA2ZTBkZWE0MGM0ZWZhYzc5MDFiNmZmZTlkNDZhNDcxOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.WVDUGY26JjZTb2-pTsz0u0ajsvrRTy6IkcmPSHZ_6SM)
> 
> hello, i got the same problem and sovled it just now.
> 
> i use this board to learn canfd. and i notice the default setting of this board is: if you use 3.3v voltage, you should provide another 5v voltage. like this: ![IMG_20231226_153342.jpg](https://private-user-images.githubusercontent.com/8655163/292827157-59de0f5e-d702-44fe-ad2c-aac8634a0f45.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwNjE3NzYsIm5iZiI6MTcwOTA2MTQ3NiwicGF0aCI6Ii84NjU1MTYzLzI5MjgyNzE1Ny01OWRlMGY1ZS1kNzAyLTQ0ZmUtYWQyYy1hYWM4NjM0YTBmNDUuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjdUMTkxNzU2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmQzMThiN2YxZjM1NzM5YjdlNWQwNmEyNDBkZjJjMjc2NjJjNWRiYjg0Y2M1YzQ0ZGU4ZmNhNTUwOWZjMWZhMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.QMPc5_xFygwuek1SEtfpPybBiFsmxD5J1m0_jbpyFII)
> 
> then it should work well.

Could you provide working code example?
Thanks!

pierremolinaro, 2024-02-28:
Your images are empty.

There are several example codes in the library, don't they suit you?

Pierre

kukumagi, 2024-02-28:
> Your images are empty. There are several example codes in the library, don't they suit you? Pierre
> […](#)

I was stuck at frame structure. But already found it.

VergilGao, 2024-02-28:
> Your images are empty. There are several example codes in the library, don't they suit you? Pierre
> […](#)

could you see the images now?

kukumagi, 2024-03-09:
I am still having trouble with the MCP2518FD Pro board.
Sometimes I get configuration error 0x1
and other times I get Configuration error 0x80000

kukumagi, 2024-03-22:
I have a problem with reading and transmitting. It wont start receiving. And the weirdest part is that if I disconnect 5v and then reconnect it somehow starts working. Do I have something wrong in my setup?

kukumagi, 2024-05-04:
I tried espressif 3.0.0-rc1 version and it works perfectly, but with version 2.x it simply does not start transmitting/receiving. Configuration works but no data comes in and unable to send frames.

kukumagi, 2024-05-25:
I dont know why but calling canfd.poll() after begin() it starts working correctly
Eg:
const uint32_t errorCode = canfd.begin(settings, [] { canfd.isr(); });
canfd.poll();