pierremolinaro/acan2517FD #44: Arduino Nano RequestedModeTimeOut

wouter-heerwegh, 2024-02-07:
Hi,

I'm trying to get the loopbackdemo sketch to run, but seem to be stuck. I'm using an mcp2517fd on a custom board with a 20MHz clock. The sketch seems to run quite well, until the requested mode is changed. It seems like the chip is stuck in configuration mode. I've tried to modify the library to stay on 800khz, but that didn't seem to have any effect.

Data transfer seems to work as expected (CS is pulled low and clock signal is sent when sending a message).
I tried modifying the interrupt routine, but it seems like it is never triggered, even though the interrupt pin is going low.

Anyone who has an idea on how I could debug this further?

pierremolinaro, 2024-02-18:
Hello,

The MCP2517FD chip is very buggy, try instead a MCP2518FD. None of tham has a RESET pin,this an actual problem, the chip cannot be reset properly in any situation.

What is your microcontroller ? Do you select ACAN2517FDSettings::OSC_20MHz or  ACAN2517FDSettings::OSC_20MHz_DIVIDED_BY_2 when you instanciate ACAN2517FDSettings ?

Pierre

wouter-heerwegh, 2024-02-19:
Hi @pierremolinaro,

I found out that my transceiver was not properly powered. After providing power to it, the mcp was able to change modes.

Thanks for the response.

closed 2024-03-28 by wouter-heerwegh