pierremolinaro/acan2517FD #48: Is the arbitration bit rate 2 Mbit/s adjustable? (MC251863 + ESP32)

dhLee325, 2024-07-26:
hello.

First of all, thank you for creating a great library.

MCP251863 + ESP32 CAN communication was carried out smoothly through the library.

We confirmed that the arbitration bit rate is 2 Mbit/s on the CAN side of the actual vehicle. However, I don't seem to be able to receive CAN data even if I set the arbitration bit rate to 2 Mbit/s. Is there another way?
(CAN communication confirmed that communication works well if matched at 1 Mbit/s)

Rio-D, 2025-08-11:
> Hi [@dhLee325](https://github.com/dhLee325),
> 
> We are also using the MCP251863. We are trying to set up a minimal example using this schematic:
> <img alt="Image" width="1264" height="771" src="https://private-user-images.githubusercontent.com/36724299/474130803-c847e648-aa68-44ce-a9a7-1cc55319c55e.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ5MzQzMTgsIm5iZiI6MTc1NDkzNDAxOCwicGF0aCI6Ii8zNjcyNDI5OS80NzQxMzA4MDMtYzg0N2U2NDgtYWE2OC00NGNlLWE5YTctMWNjNTUzMTljNTVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA4MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwODExVDE3NDAxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI4ZTQ2MmU3MWQwYmViNWEwODVkZmNhMzhjZDZlZmZkZTgyNzQwNjRjMTI3YzE0NGQ3NzczOTY2NmU0MWMzMWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0ZK0DlOv6qOP6GKS1UNRWRQ4aPwRf7lSFh714z1gI2Q">
> 
> the STBY is on LOW and we are able to execute the "Loopbackexample" and receive successfull returns. However, when entering normal mode or External loopback we do not observe data on the Bus.
> 
> It would be a real help if you'd share your schematics and Code so we can have a look at what we might've done wrong.
> 
> Greetings, JDF

While in theory a 20Mhz oscillator is enough to handle 2Mbit/s, it will only work in ideal conditions; i.e short bus with not much noise.
If you want things to work with 2Mbit/s and up you need to use a 40MHz oscillator. 

They are hard to come by, so you will have to make your own most likely. Something I am working on but don't have much time.

pierremolinaro, 2025-08-11:
> Rio-D

No, it is not possible. CANFD standard limits the arbitration bit rate to 1 Mbit/s. However, the data bit rate can be x10 more faster.

Pierre