pierremolinaro/acan2517FD #53: Does the library support 2x 2518FD chips on same board?

dalathegreat, 2024-11-28:
Working with some custom PCBs that have 2x MCP2518 chips on the same SPI

```
// CAN_FD the first defines
#define MCP2517_SCK 17  // SCK input of MCP2517
#define MCP2517_SDI 23  // SDI input of MCP2517
#define MCP2517_SDO 39  // SDO output of MCP2517
#define MCP2517_CS 21   // CS input of MCP2517 //21 or 22
#define MCP2517_INT 34  // INT output of MCP2517 //34 or 35

// CAN_FD the 2nd defines
#define 2nd_MCP2517_SCK 17  // SCK input of MCP2517
#define 2nd_MCP2517_SDI 23  // SDI input of MCP2517
#define 2nd_MCP2517_SDO 39  // SDO output of MCP2517
#define 2nd_MCP2517_CS 22   // CS input of MCP2517 //21 or 22
#define 2nd_MCP2517_INT 35  // INT output of MCP2517 //34 or 35
```

Has anyone been able to use this library with two chips at the same time? One of the chips start, but the second fails to start with errorCode 1

dalathegreat, 2025-01-18:
Some update on this, [I got the boards to initialize properly](https://github.com/dalathegreat/Battery-Emulator/pull/800), but after running for a  few seconds I get send_ok going bad, and buffer overflows.

tomtom0707, 2025-01-23:
Hi, I would be interested in the topic just as much.
See the test program below, which is running. Is this the right way to use the library for 2 x MCP2518FD? 
What I have to do to give the 2nd controller different settings? I'm not a master of programming...
Note, I left out the speed limit ;)

`//------------------------------------------------------------------------------
//  ACAN2517FD Driver object
//------------------------------------------------------------------------------

SPIClass vspi (VSPI) ; // See ESP32 demo sketch: SPI_Multiple_Busses

ACAN2517FD can0 (MCP2517_CS0, vspi, MCP2517_INT0) ;
ACAN2517FD can1 (MCP2517_CS1, vspi, MCP2517_INT1) ;

ACAN2517FD *can[2];

CANFDMessage frame ;
uint8_t i;

//------------------------------------------------------------------------------
//   SETUP
//------------------------------------------------------------------------------
void setup () {
//--- Start serial
 Serial.begin (115200) ;
 while (!Serial) {
    delay (50) ;
 }
 can[0] = &can0;
 can[1] = &can1;
//----------------------------------- Begin SPI
 vspi.begin (MCP2517_SCK, MCP2517_MISO, MCP2517_MOSI) ;
//---------------------- CAN0 ----------------------------------------------------
  Serial.print("sizeof (ACAN2517FDSettings): "); Serial.print(sizeof (ACAN2517FDSettings));
  Serial.println(" bytes"); Serial.println("Configure CAN0:");

  ACAN2517FDSettings settings (ACAN2517FDSettings::OSC_4MHz10xPLL, 1000 * 1000, DataBitRateFactor::x8) ;
  settings.mRequestedMode = ACAN2517FDSettings::InternalLoopBack ; // Select loopback mode
  uint32_t errorCode[2]={0};
 
  errorCode[0] = can[0]->begin (settings, [] { can[0]->isr () ; }) ;
  if (errorCode[0] == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Arbitration Phase segment 1: ") ;
    Serial.println (settings.mArbitrationPhaseSegment1) ;
    Serial.print ("Arbitration Phase segment 2: ") ;
    Serial.println (settings.mArbitrationPhaseSegment2) ;
    Serial.print ("Arbitration SJW:") ;
    Serial.println (settings.mArbitrationSJW) ;
    Serial.print ("Actual Arbitration Bit Rate: ") ;
    Serial.print (settings.actualArbitrationBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact Arbitration Bit Rate ? ") ;
    Serial.println (settings.exactArbitrationBitRate () ? "yes" : "no") ;
    Serial.print ("Arbitration Sample point: ") ;
    Serial.print (settings.arbitrationSamplePointFromBitStart ()) ;
    Serial.println ("%") ;
  }else{
    Serial.print ("ConfigCAN0 error 0x") ;
    Serial.println (errorCode[0], HEX) ;
  }
// ---------------------- CAN1 ----------------------------------------------------

	//Config CAN1 - settings ???

  errorCode[1] = can[1]->begin (settings, [] { can[1]->isr () ; }) ;
  if (errorCode[1] == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Arbitration Phase segment 1: ") ;
    Serial.println (settings.mArbitrationPhaseSegment1) ;
    Serial.print ("Arbitration Phase segment 2: ") ;
    Serial.println (settings.mArbitrationPhaseSegment2) ;
    Serial.print ("Arbitration SJW:") ;
    Serial.println (settings.mArbitrationSJW) ;
    Serial.print ("Actual Arbitration Bit Rate: ") ;
    Serial.print (settings.actualArbitrationBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact Arbitration Bit Rate ? ") ;
    Serial.println (settings.exactArbitrationBitRate () ? "yes" : "no") ;
    Serial.print ("Arbitration Sample point: ") ;
    Serial.print (settings.arbitrationSamplePointFromBitStart ()) ;
    Serial.println ("%") ;
  }else{
    Serial.print ("ConfigCAN1 error 0x") ;
    Serial.println (errorCode[1], HEX) ;
  }
  Serial.println ("los gehts...");
}
//------------------------------------------------------------------------------
//   LOOP
//------------------------------------------------------------------------------
static uint32_t gReceivedFrameCount[2] = {0};
static uint32_t gSentFrameCount[2] = {0};

static const uint32_t MESSAGE_COUNT = 10000 ;
uint8_t	stopp[2] = {0};
//------------------------------------------------------------------------------

void loop () {

  for(i=0;i<2;i++)
  {
     if (gSentFrameCount[i] < MESSAGE_COUNT)
     {
      const bool ok = can[i]->tryToSend (frame) ;
      if (ok) {
       gSentFrameCount[i]++;
      }
     }
     else
     {
     	if(stopp[i] == 0){
     		stopp[i] = 1;
     	  Serial.print ("CAN"); Serial.print (i);
     	  Serial.print (" Sent: ") ;
        Serial.print (gSentFrameCount[i]) ;
        Serial.print ("; Received: ") ;
        Serial.println (gReceivedFrameCount[i]) ;
      }     
     } 
  } 

  for(i=0;i<2;i++)
  {
   if (can[i]->available())
   {
    CANFDMessage frame ;
    can[i]->receive(frame) ;
    gReceivedFrameCount[i] ++ ;
   }
  }
}
`

tomtom0707, 2025-01-25:
If you do this with an ESP32, pay attention to the selection of the INT pins. Two edge-triggered pins in the same group can cause problems here (in your example 34 and 35).
see: https://docs.espressif.com/projects/esp-chip-errata/en/latest/esp32/03-errata-description/esp32/gpio-edge-interrupts.html

I tested it with ESP32 and two MCP2518.
Test InternalLoopBack, also different settings - ok.
Test Normal20B same settings one CAN bus - ok.
Test Normal20B different settings two CAN buses (only send) - ok.
So I think it works with two MCP2518, but I haven't tested all the variants.
I'm from Germany, I use the loop without speed limit (delay), no errors occur.
General note: If you really want to send quickly on the bus, you have to increase the send buffer (mControllerTransmitFIFOSize).

pierremolinaro, 2025-01-27:
Hi,

You can use two ACAN2517FDSettings objects :

//--- First one
  ACAN2517FDSettings settings0 (ACAN2517FDSettings::OSC_4MHz10xPLL, 1000 * 1000, DataBitRateFactor::x8) ;
  settings0.mRequestedMode = ACAN2517FDSettings::InternalLoopBack ; // Select loopback mode
  uint32_t errorCode[2]={0};

  errorCode[0] = can[0]->begin (settings0, [] { can[0]->isr () ; }) ;
   ...
//--- Second one
  ACAN2517FDSettings settings1 (ACAN2517FDSettings::OSC_4MHz10xPLL, 1000 * 1000, DataBitRateFactor::x8) ;
  settings1.mRequestedMode = ACAN2517FDSettings::InternalLoopBack ; // Select loopback mode
  uint32_t errorCode[2]={0};

  errorCode[1] = can[1]->begin (settings1, [] { can[1]->isr () ; }) ;
  
settings0 and settings1 can have different settings.

Pierre

dalathegreat, 2025-01-28:
Thanks @pierremolinaro , we tried this but it did not help. After 3-5 minutes of successful operation of two CAN-FD channels at once, the CAN buses go down. Seems like some timing issue or something, hard to say :/

rjsc, 2025-08-21:
`#ifdef ARDUINO_ARCH_ESP32`
`        attachInterrupt (itPin, inInterruptServiceRoutine, FALLING) ;`
`      #else`
`        mSPI.usingInterrupt (itPin) ; // usingInterrupt is not implemented in Arduino ESP32`
`        attachInterrupt (itPin, inInterruptServiceRoutine, LOW) ; // Thank to Flole998`
`      #endif`

I noticed that for ESP32 a falling edge interrupt is setup, while for non-ESP32 platforms a LOW Level Triggered interrupt is enough to make the code work.

Since the ESP32 has a bug with edge triggered interrupts as mentioned above by tomtom0707, and in the errata is state that Level triggered interrupts work, cant we also attach a LOW Level trigerred interrupt for ESP32 like so?

 `#ifdef ARDUINO_ARCH_ESP32`
`        attachInterrupt (itPin, inInterruptServiceRoutine, LOW) ;`
`      #else`
`        mSPI.usingInterrupt (itPin) ; // usingInterrupt is not implemented in Arduino ESP32`
`        attachInterrupt (itPin, inInterruptServiceRoutine, LOW) ; // Thank to Flole998`
`      #endif`